RoganDawes/P4wnP1 #184: Add keyboard language

TheMMcOfficial, 2017-12-26:
Hi,
I'm new with the P4nwP1. I have tested the backdoor option and it work on my laptop but not on my Desktop. I don't have the same keybord. My laptop have the French Canadian and my Desktop have the Multilang Canandian. I don't know if it's possible to add this new keyboard to the scripts. I don't know where I can do it...

Swiftb0y, 2017-12-27:
You can create it yourself #175

TheMMcOfficial, 2017-12-28:
Thanks, I have create my version for the keyboard Canadian Multilangual. I can upload it if you want it. It is not perfect I do not have put all the pound symbols... but it works!

Swiftb0y, 2017-12-28:
If you could make sure that all every level 1 and 2 Kedboard chars work, we would be glad if you’d open a Pull request.

TheMMcOfficial, 2017-12-29:
I got all the chars from a-z, A-Z, 0-9, and all the special chars made with the shift key and the alt key!

Swiftb0y, 2017-12-29:
Thats good. I think it’s ready for a PR to the [duckencoder repo](https://github.com/mame82/duckencoder).

TheMMcOfficial, 2017-12-29:
Your link doesn't work...

Swiftb0y, 2017-12-30:
Sorry I forgot the .py
https://github.com/mame82/duckencoder.py

TheMMcOfficial, 2018-01-09:
I have create my pull request there: https://github.com/mame82/duckencoder.py/pull/7

closed 2018-08-13 by TheMMcOfficial

Msprg, 2019-01-04:
Hi, @TheMMcOfficial I would like to add my own keyboard language (slovak) as well, but I am not sure, how to do it. I does not know wich part do i have to change in the keyboard layout file, or where to get the ascii codes from. Could you please sort of guide me through it? If it does not bothers you, of course.

Swiftb0y, 2019-01-05:
This repository is outdated and unmaintained. Please use the newer [P4wnP1 A.L.O.A](https://github.com/mame82/p4wnp1_aloa). 

This is a snippet of a guide on writing Keyboard layouts for P4wnP1 A.L.O.A (**NOT THE LEGACY P4WNP1 (THIS)**):

# Adding a Keyboard layout.

All layouts reside in [dist/keymaps/](https://github.com/mame82/P4wnP1_aloa/tree/master/dist/keymaps/)  
They are huge JSON files which define a dictionary which enable an easy lookup for chars that can be easily typed by the keyboard.  
Each JSON file consists of a 'header' with metadata and a body with the dictionary.  
## Header syntax
```json
"Name": "<LANG>",
"Description": "<a one-sentence description of the mapping>"
```
## Body Syntax
Each character entry consists of one or more keys or modifiers which indicate which keys have to pressed on the 'virtual keyboard' for the character to be typed.
```json
"Mapping": {
  "<character>": [{"Modifiers": ["<Modifiers>"], "Keys": ["<Keys>"]}],
  etc.
}
```
Most the names of the Modifiers and Keys are self-explanatory or can be found in common.json. If you are looking for a key that is not contained in common.json, you should be able to find it in the keyboard_globals.go  

## Examples
german keyboard:
```json
"a": [{"Modifiers": [],	"Keys": ["KEY_A"]}],
"A": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_A"]}],
```

## Special cases
Sometimes multiple different keys, typed in sequence are needed to type a character, this can easily be achieved by using multiple array elements within a character entry: (german example)  
```json
"^": [{"Modifiers": [], "Keys": ["KEY_GRAVE"]}, {"Modifiers": [], "Keys": ["KEY_SPACE"]}],
```