RoganDawes/P4wnP1 #185: help wanted with a tool

hahnstep, 2017-12-27:
Hi 

https://github.com/hahnstep/bunnyscript

i created a tool for easy payload creation everything works fine, except one thing. 
every time if i login i got this error 

-bash: LED: command not found

LED is a shell function from my tool and is sourced by payload and not by P4wnP1 framework !

some body an idea where i can search for this glitch ?

Swiftb0y, 2017-12-27:
I’m not familiar enough with the fine details of bash and I can’t test it right now but don’t you have to use the `function <name>` keyword to define funktions?

hahnstep, 2017-12-27:
I found the problem, the payload function "onLogin" is a part of the profile in /tmp/profile 

So everything works fine, dont use bunnyscript functions inside the payload function "onLogin".

thanks

closed 2017-12-27 by hahnstep