RoganDawes/P4wnP1 #186: Changing P4wnPI SSID and Passphrase

KarlesP, 2017-12-27:
So I installed the P4wnPI correctly on my RPI0W and the thing is that when I log in to my pi via ssh (not as root) I open up  ~/P4wnP1/setup.cfg using "sudo nano", then I go to  WIFI_ACCESS_POINT_NAME and WIFI_ACCESS_POINT_PSK, change their values into something else, then I reboot my RPI and still the access point is shown as P4wnP1. Am I missing something?

hahnstep, 2017-12-27:
HI check the payload does the payload have a WIFI_ACCESS_POINT configuration ?

KarlesP, 2017-12-27:
@mame82 Yes it does, but it takes your default, should I change the network.txt?

hahnstep, 2017-12-27:
Hi the payload overwrite the configuration from setup.cfg

mame82, 2017-12-27:
In case an option like `WIFI_ACCESS_POINT_NAME` is set in a payload, it has priority ofer the default option in `setup.cfg`. 

So you have to either change the options in the selected payload or delete them from the payload (results in fall back to the oprion from setup.cfg).

For example deleting `lang` from all keyboard payloads would always use the setting of setup.cfg. Specifying a `lang` value in every keyboard payload would allow to use a fifferent keyboard language per payload.

Same goed true for every other option available

KarlesP, 2017-12-27:
@mame82 I read that from your doc (I think) and I changed it, thank you for your support.

closed 2017-12-30 by mame82