RoganDawes/P4wnP1 #188: Downloading Files Issue (hid_backdoor.txt)

SQLisLOVE, 2018-01-09:
Hi,
    So I've been playing around with MaMe's project and I absolutely love it. For a while my main focus in coding was creating bat. scripts to export files from a victim PC. I created a bat. file that copied pictures from the victims PC and saved them to the HACKIN9 drive (xcopy "%USERPROFILE%\pictures"). Yes, I am aware that script isn't complete but I didn't feel like copying it from my pi w. So i would run the script and I would get a return error saying that I could not create files to the drive. I don't know why this is happening and it has me stumped. I don't want to create the server and website for the hid_backdoor_remote.txt because i don't want to control it from the internet. So I cannot get the download command working. If you can contribute anything I would much appreciate it.

mame82, 2018-01-14:
The `download` command of the backdoor save files from the remote folder (shown with `pwd` when backdoor is connected) to P4wnP1's file system current dir (shown with `lpwd` when backdoored client is connected).
The hakin9 payload uses USB mass storage, on the other hand. Files written to the mass storage end up in the image backing the mass sorage device (and thus not directly on P4wnP1's file system).

SQLisLOVE, 2018-01-20:
So, I downloaded a file from the victim pc right? It is stored in the backdoor directory. How can I move the file to the USB_STORAGE to actually view it. Thanks MaMe and I love the project.

Swiftb0y, 2018-01-20:
While it is possible, the USB storage isn’t necessarily meant to be used transferring files between P4wnP1 and your Workstation. If you only need to get some files from the Pi, you can start a quick webserver in the directory you want to download the files from by doing a `python3 -m “http.server”`.  If you need bidirectional file transfer you could mount the USB_STORAGE/Image.bin as a loopback device. 
Unfortunately if the storage-gadget functionality is used, mounting the image simultaneously as a loopback device will result in inconsistent data which will probably corrupt your files.

mame82, 2018-04-07:
The `download` command of the hid backdoor stores file to:
a) an absolute target path on P4wnP1, if given as 2nd argument to the command
b) if no target path is provided, file are stored to P4wnP1's current working dir, which could be shown with `lpwd` and changed with `lcd` (`pwd` and `cd` change the directory on the victim)

closed 2018-04-07 by mame82