RoganDawes/P4wnP1 #193: Connect the Raspberry pi zero to internet

Anonyvius, 2018-01-14:
Hey, My raspberry is fully working and P4wnP1 is running. But i wanted to download some github files and because the creates his own network he cant connect to the internet anymore. So my question is if there is a way to download some files via internet. Or add a connect the raspberry to the internet AND let you ssh into it.

I hope you unterstand what i have wirtten.

Swiftb0y, 2018-01-17:
I recommend you set up your Pi as a WiFi client. This way, it can just simply connect to your WPA2 Router at home to get access to the internet and still use another device in your home network to interact with the Pi.
The relevant settings in the setup.cfg/payload start with the WIFI_CLIENT prefix.

LmTClever, 2018-01-26:
Hey Swiftboy, how would I do that? Can you provide some steps if possible?

Swiftb0y, 2018-01-26:
I don’t have time write a complete step by step guide so I hope the [wiki](http://p4wnp1.readthedocs.io/en/latest/P4wnP1-W/Wifi-Client/) can answer most of your future questions.

hahnstep, 2018-01-26:
try this payload, change YourAPName and YourAPPassword to your home wifi 

USB_VID="0x1D6B"        # Vendor ID
USB_PID="0x0237"        # Product ID

USE_ECM=true            # kept enabled as fallback (could allow forwarding connection trough WiFi if added to payload)
USE_RNDIS=true          # kept enabled as fallback (could allow forwarding connection trough WiFi if added to payload)
USE_HID=false
USE_RAWHID=false        
USE_UMS=false

# disable setting of static routes for all IPv4 addresses
ROUTE_SPOOF=false

WIFI_ACCESSPOINT=false

WIFI_CLIENT=true
WIFI_CLIENT_SSID="YourAPName" # name of target network
WIFI_CLIENT_PSK="YourAPPassword" # passphrase for target network

closed 2018-04-07 by mame82