RoganDawes/P4wnP1 #195: HID_keyboard payload doesn't work (nothing happens)

isecuritypro, 2018-01-18:
Hi 

I have installed P4wnP1 in proper way there is nothing wrong, i have uncommented the hid_keyboard module from the setup.cfg file and i have rebooted my raspberry pi zero and for some reason it the module doesn't work (nothing happens)

Can you help me to solve the problem ???

Swiftb0y, 2018-01-18:
To make it easier for us, please refer to the [troubleshooting guide](http://p4wnp1.readthedocs.io/en/latest/Troubleshooting/).

HitItAll, 2018-01-18:
1. Make sure your cable has data pins (you can check to see if the computer
can access your phone through this cable)

2. The RasPi0 has 2 USB ports, make sure you are using the one marked USB
(not PWR)

3. SSH into the P4wnP1 and try to run: python
~/P4wnP1/payloads/hidtools/P4wnP1.py (I'm not sure if that is the exact
location of the file but...)

Good Luck

Swiftb0y, 2018-01-18:
> 3. SSH into the P4wnP1 and try to run: python
~/P4wnP1/payloads/hidtools/P4wnP1.py (I'm not sure if that is the exact
location of the file but...)

The exact location would be hidtools/backdoor/P4wnP1.py in your P4wnp1 base directory.

HitItAll, 2018-01-19:
>    1. SSH into the P4wnP1 and try to run: python

Thanks for correcting

Mikhail5555, 2018-02-01:
I was experiencing the same problem, when I manually run P4wnP1.py I get the following:
```shell
pi@MAME82-P4WNP1:~ $ python P4wnP1/hidtools/backdoor/P4wnP1.py
Traceback (most recent call last):
  File "P4wnP1/hidtools/backdoor/P4wnP1.py", line 1325, in <module>
    HIDin_file = open(dev_file_in_path, "rb")
IOError: [Errno 2] No such file or directory: '/dev/hidg1'
Cleaning Up...
Traceback (most recent call last):
  File "P4wnP1/hidtools/backdoor/P4wnP1.py", line 1361, in <module>
    ll.stop() # send stop event to read and write loop of link layer
NameError: name 'll' is not defined
pi@MAME82-P4WNP1:~ $
```

Swiftb0y, 2018-02-01:
`IOError: [Errno 2] No such file or directory: '/dev/hidg1'` suggests that the Keyboard isn’t setup correctly since the corresponding file for interfacing doesn’t exist.
Make sure that the settings in the payload are correct and post the output of `sudo journalctl -u P4wnP1.service` once you made sure setup.cfg and hid_backdoor.txt  are correct.

Mikhail5555, 2018-02-01:
Thanks for your suggestion, I was kinda experiencing more problems so I reinstalled everything and it works now.
Maybe a good thing to note: I clean installed "Stretch Lite image" Twice and it didn't work any of those times, now I did "Jessie Lite image" and it worked instantly.

Swiftb0y, 2018-02-02:
Good to know. They changed some stuff in stretch but mame made sure P4wnP1 works nonetheless. If you have the nerves to experiment a little in the future, try it again with stretch and post some of the logs. My installation on stretch works flawlessly.

closed 2018-04-07 by mame82