RoganDawes/P4wnP1 #198: [Not an issue] Faking the size of the bin image used for USB storage

leperjulien, 2018-01-29:
Hi there,

I've been playing around with p4wnP1 for a while now and its a pretty amazing tool !

I was wondering if you think it would be possible to fake the size of the USB Storage bin file to display something like 1 TO or more using something like this : 

https://superuser.com/questions/270194/how-is-the-capacity-of-a-harddisk-faked

https://www.youtube.com/watch?v=pHQxB7Ve2wQ

I might try to play with it and see if i can manage to make it work but first i would like to know if that's even possible ?

Thanks

HitItAll, 2018-02-04:
From what I understand this is not possible, I am not an expert but believe
that it was the firmware of these USBs that believed they were 1TB. All the
P4wnP1 does is give the PC access to the .img  file. Again I AM NOT SURE.

leperjulien, 2018-02-04:
@HitItAll I tought too at first but then i kept reading about it , and it seems that it's possible witouth having to use modified firmware :)

This guy is doing it : 
`
| Here is how you create a 1,000,000,000,000 bytes (1 TB) disk on key (using Linux):   
Create fake formatted 1 TB disk:  mkdosfs -C  temp_file 1000000000  
Check that it really happened  ls -lh temp_file  
Connect your DoK and check how it was mounted:  mount  
Find the device name, something like /dev/sdb1 (If you make a mistake here, you might ruin some other disk connected to the system, so be careful !) 
Unmount it:  sudo umount /dev/sdb1  
Find out the size of our FAT table:  ls -s temp_file    (The first number is the size in kilobytes)   Copy over to DoK  echo "head -c [size_from_above]K temp_file > /dev/sdb1" \| sudo sh  
Freak out your friend!You can mount it locally for playing around like this:  
mkdir test
sudo mount temp_file test -o loop
-- | --

`

And i noticed inside the install.sh script of p4wnp1 it's also using mkdosfs command to create the bin file used for USB_STORAGE , and since this technique is also using mkdosfs maybe something can be made here ;) But like you I AM NOT SURE

If anyone could confirm that would allow me to start experimenting (or not)

HitItAll, 2018-02-05:
Why don't you try? Worst comes to worst you just do a fresh install