RoganDawes/P4wnP1 #199: Always boots to Interactive mode

migshin, 2018-02-05:
Hello,

I have recently had an issue with P4wnP1 in which it always boots into interactive mode, regardless of the circumstances. I've tried to use many different cables on many different computers, and it still doesn't work. I even re-installed P4wnP1 a couple times, but to no avail.

I saw a previous issue of the same nature that you resolved by setting OTG mode to always be false in the "https://github.com/mame82/P4wnP1/blob/master/boot/boot_P4wnP1#L60" folder. I couldn't find this folder, but I did find the "detect_usb_hostmode" function in /P4wnP1/boot/init_usb.sh and changed it so that the OTG_MODE=false in both the "if" and "else" parts of the function. This didn't work either, and my pi still boots into interactive mode.

Is there any command that forces the program out of interactive mode, or is it possible that my raspberry pi is defective?

Thanks.

bpartin2009, 2018-02-06:
If you look on your raspberry pi in the P4wnP1/boot/ folder you will see a file called boot_P4wnP1 and the line number is at 115 that has the detect_OTG_mode code in it. If on line 116 you put OTG_MODE=false, then it should never go into interactive mode.

migshin, 2018-02-07:
That seemed to have worked. For whatever reason, the file wasn't showing up when I used the "ls" command, but this time I just typed "nano boot_P4wnP1" and it put me in the file.

Thank you for your time.

AlCon1989, 2018-02-08:
Hello migshin,
I also added OTG_MODE=false at line 116. And it does indeed boot up in usb host mode, however my collected directory is empty. It will not collect password hashes from either windows 7 8 or 10. 
Did you get LockPicker to work ? If yes, how ? I reinstalled jessie 3 or 4 times i reinstalled stretch 3 or 4 times, changed usb cables enabled otg_mode to false and still it LockPicker never works.
Looking forward.

migshin, 2018-02-09:
AlCon1989,
Lockpicker hasn't worked for me either, but I don't believe it has anything to do with OTG mode or P4wnP1 updates; Microsoft simply patched the exploit. You can read more about it here: https://github.com/mame82/P4wnP1/issues/158

mame82, 2018-02-12:
The issue allowing LockPicker to work was reported to Microsoft and fixed back in November 2017.

Don't rely on the exact KB number given by myself, as the patches are cumulative.

The repo contains a writeup which gives details on the LockPicker payload and why it shouldn't work anymore on a healthy system

closed 2018-04-07 by mame82