RoganDawes/P4wnP1 #200: Problem connecting to respiZero 1.3 via usb ssh on win10/Ubuntu

Fireflaker, 2018-02-07:
Hi,
I am trying to connect the pi to my win10 computer via USB ssh without using any network adapters. It does not work. Here is the problem. 
I followed the tutorial adding code to config.txt and  cmdline.txt. I also created an empty file named "ssh". Then I connected I to my win10 computer and tried to ssh into it using putty. Failed.
I tried to update the driver on win10 device manager but the pi is registered as a COM port device instead of an unknown device. This prevents me from changing its driver to “Remote NDIS Compatible Device” because there is no such option. (http://www.circuitbasics.com/raspberry-pi-zero-ethernet-gadget/) And I am stuck. I tried to use Ubuntu in VirtualBox but what all I get is 
[image]
(same is Linux)

Any suggestions? Thanks...(Now I am starting to hate Windows)

Fireflaker, 2018-02-07:
[image]
I tried installing the driver I downloaded separately but it does not work.

Fireflaker, 2018-02-07:
[image]
Also, unable to ping. 
I tried the same pi on a different win10 computer and the result is the same

ukscone, 2018-02-07:
try changing using the info from "Making it work on Windows" section from http://irq5.io/2016/12/22/raspberry-pi-zero-as-multiple-usb-gadgets/ 
I was having problems with getting my pizero to show up as RNDIS on win10 until i "borrowed" the setup from that page.

ukscone, 2018-02-07:
I think you need to add something like
`echo 0xEF > bDeviceClass
echo 0x02 > bDeviceSubClass
echo 0x01 > bDeviceProtocol`

to setup.cfg

ukscone, 2018-02-07:
it probably doesn't go in setup.cfg but it'll go somewhere :) i'll have a dig around and see if i can find where

ukscone, 2018-02-07:
weird it looks like it's already set like that in /boot/init_usb.sh so it should be working

Fireflaker, 2018-02-07:
I tried plugging in the pi without having the SD card and it is registered as unknown device and I can install 
 the right driver for it. However, the driver installation fails and nothing happens. I am using many pi zeros and there should not be a hardware problem. 
Thank you very much for your help

Fireflaker, 2018-02-07:
https://answers.microsoft.com/en-us/windows/forum/windows_10-networking-winpc/windows-10-vs-remote-ndis-ethernet-usbgadget-not/cb30520a-753c-4219-b908-ad3d45590447
seems to be a common issue.

ukscone, 2018-02-07:
sdcardless booting of a pizero only works on linux and mac afaik as it only identifies itself as an ECM device not RNDIS. 

yes it's a common problem and a right PITA :( :) I managed to get my pizeros working how I wanted on windows using the info from several sites and ended up making this gist https://gist.github.com/ukscone/d2e4fa7aa3b95ce6cdcc7cbffe2e4453 i don't know how well it'll translate to P4wPi as i haven't yet tried P4wPi yet as i got sidetracked from my plans but i'll have a proper fiddle tonight

Fireflaker, 2018-02-07:
http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/?os=Windows&os-version=10
I am following this tutorial and it seems to be helpful/

mame82, 2018-02-07:
The problem that Raspbian's USB RNDIS gadget is misdetected as serial COM port on Win10 is well known. 

P4wnP1 doesn't face this issue, once it is installed RNDIS is working correctly (done by adding correct OS descriptors which make Windows enumerate RNDIS correctly).

The issue you have to deal with, is to prepare an Internet connected Pi0, in order to be able to install P4wnP1 at all.

I gave instructions on several ways to achieve this, and the problem you're facing is the exact reason for excluding Win10 USB connections from these setup instructions.

The easiest way to get a bare Raspbian (no P4wnP1 installed) Pi0W connected to the Internet is the one described by "Dan the IOT man" which is referenced in the readme.

In case you use a Pi0 without WiFi capabilities, I suggest following my instructions for a headless installation supported by a Kali Linux host.

Once P4wnP1 is installed, RNDIS is working fine on Win10

Fireflaker, 2018-02-08:
I bought a USB LAN adapter and a USB Wifi adapter. They are on the way. Just asking though, is it possible to install a full out Linux system and get it to SSH via USB? Or would it be too complicated to be worthy of trying?

Fireflaker, 2018-03-02:
Now I can ssh into it via an ethernet cable but I can not share the internet connection of my computer with the raspberry pi. The computer is also running a DHCP server. What should I do? Thanks.

mame82, 2018-04-07:
Please use the ready-to-go image provided here https://github.com/mame82/P4wnP1/releases

ICS tutorial for windows is here:
https://www.youtube.com/watch?v=QEWaIoal5qU&t=17s

closed 2018-04-07 by mame82