RoganDawes/P4wnP1 #201: can't usb ssh Mac

pythlang, 2018-02-11:
I previously had this working no problem and accidentally flashed my SD for another project

I'm trying to ssh to p4wnp1 via 172.16.0.1 but I'm unable as the ip is set to 169.xxx and although I am able to gain access to the pi if I manually change the ip to 172.16.0.2 and the subnet to 255.255.255.252, I still cannot access the outside internet from the pi

I can connect to 172.24.0.1 no problem.

I tried following the youtube video mame82 uploaded but I'm unable to continue as I can't set the network from public to private on a mac using Get-Net....

How in the h*ll did it configure this originally?

My steps:

flash
drop ssh & wpa_supplicant files
edit config.txt -> dtoverlay=dwc2
edit cmdline.txt -> add that string after root wait & before quiet
insert sd to raspberry
use microusb instead of power
insert raspberry pi
rndis appears
share internet connection with it
sudo apt-get update
follow install actions on READme
no error messages
sudo shutdown -h now
unplug/replug -> pi boots
that's where my problems begin

any solutions/hints? I'm sure a more experienced networker will figure this out no problem, but i'm barely beginning to understand how they interact.

Thanks again

pythlang, 2018-02-11:
Everything seems to work fine, meaning I can ssh into 172.16.0.1, until I start internet sharing with P4wnP1; I either cannot login to p4wnp1 to 172.xxx after enabling internet or the ssh session drops while toggling internet sharing and the DHCP reverts to 169.xxx

mame82, 2018-02-12:
The IP gets overwritten with an APIPA IP after enabeling ICS, by Windows. That's normal behavior.

Here's a tutorial on how to handle this and connect P4wnP1 to Internet using ICS:

https://youtu.be/QEWaIoal5qU

pythlang, 2018-02-12:
Bro, I'm using a Mac, not Windows but maybe something similar is happening.

mame82, 2018-02-12:
I'm sure that apple implemented APIPA, too ;-)

pythlang, 2018-02-12:
That does indeed appear what is happening, but I'm unsure of how to convert my profile from public to private. Do you know any workarounds?

pythlang, 2018-02-12:
Now I'm getting this
[image]

HitItAll, 2018-02-12:
This is most likely because P4wnP1 is taking that IP. Try setting your IP
to 172.16.0.2.
Good luck!

pythlang, 2018-02-12:
I meant to set it to 172.16.0.2, but I'm still having the same problem unfortunately.

mame82, 2018-04-07:
Please try the ready-to-go image provided here https://github.com/mame82/P4wnP1/releases