RoganDawes/P4wnP1 #203: [NOT AN ISSUE] Uninstall P4wnP1 or connect wlan0 to an other network?

m0rpheus1337, 2018-02-15:
Hello, 

I've got a raspberry pi 0 W and I've installed P4wnP1 but without "--recursive" option.
So I delete P4wnP1 directory but my wlan0 interface is not found by raspbian (on graphical interface).
Wlan0 is up in terminal but it don't connect on my network, even with wpa_supplicant.

P4wnP1 may have changed my wireless interface configuration. 
Can you help me to uninstall completely P4wnP1 without internet connection or repare my interface ?

Thanks ! :)

TheDevFreak, 2018-02-15:
You could just re-image the card?

m0rpheus1337, 2018-02-15:
Yes I can re-image the card but I've installed packages and so if I can properly delete P4wnpi, it's better.

But if there isn't possibility, I'll do that ! 
Thanks for your answer !

HitItAll, 2018-03-18:
My answer is very late but no, you can't uninstall P4wnP1 (other than by
re-imaging it)

mame82, 2018-04-07:
Please use the ready-to-go image provided here https://github.com/mame82/P4wnP1/releases

closed 2018-04-07 by mame82

Pomurphy, 2018-06-13:
Hi.
It is rather easy to do so.
You should take the installer 
https://github.com/mame82/P4wnP1/blob/master/install.sh
and revert  the actions,
In short : enable the disabled services and disable P4wnP1 services.. 

Regards