RoganDawes/P4wnP1 #204: error during installation

ikuamike, 2018-02-18:
error appears after 
Unpacking John the Ripper Jumbo edition...
error: pathspec 'jtr_stretch' did not match any file(s) known to git.

Swiftb0y, 2018-02-18:
http://p4wnp1.readthedocs.io/en/latest/Troubleshooting/#hid_backdoor-and-hid_frontdoor-are-not-working

ikuamike, 2018-02-18:
I used --recursive flag, also during cloning it hangs at the end so I use ctrl ^C to stop and checking using 
git checkout -f HEAD
response is
Your branch is up-to-date with 'origin/master'.
but nothing is in the john the ripper folder

mame82, 2018-02-18:
Yes, you shoul wait

ikuamike, 2018-02-18:
[image]

An error occurred during cloning there but it resolved itself and finished okay. 
Installation ended okay as well.
Thanks for the help.

closed 2018-02-18 by ikuamike