RoganDawes/P4wnP1 #206: [NOT A ISSUE] Idea: Add OS recognition

gloglas, 2018-02-26:
Can you please add some thing like this?

(Arduino library) https://github.com/keyboardio/FingerprintUSBHost

jcstill, 2018-03-09:
I have been looking into this ever since I saw this post a couple days ago. (Its not a high priority for me so I've been pushing it off)

### Over Network
When the `PAYLOAD=network_only.txt` is enabled in `setup.cfg`, the device is in a mode that emulates an "Ethernet over USB" connection with the default gateway being `172.16.0.1`.

**Use nmap**
`sudo nmap -sn 172.16.0.* | grep 172` 
shows a simplified list the devices on the network. There should only be 2 (the pi, and the target)
`sudo nmap -O 172.16.0.(whatever ip the target is)`
can be used to determine the OS.

This method is one of many ways to fingerprint an OS via a network. Others include Xprobe2 and p0f.
See [here](https://www.sans.org/reading-room/whitepapers/authentication/os-application-fingerprinting-techniques-32923)

### Over USB
**USB fingerprinting**
This method may be able to identify a target with more accuracy. I have been looking at these articles:

> "demonstrate [...] that we can differentiate not only between operating systems, but between seemingly unnoticeable differences in machine model types as well. We also show that we can differentiate between real and virtualized hosts responding to USB stimuli"

Quote from [here](https://pdfs.semanticscholar.org/152d/ebadbbeb1322be2793f5257aabf8e3237356.pdf)
Similar article [here](https://cise.ufl.edu/~butler/pubs/ndss14.pdf)

### Tool that does both:
I will started a [repo](https://github.com/jcstill/P4wnP1_OS_Recognition) for this based on the links above, so if anyone wants to help I would appreciate it greatly.

mame82, 2018-03-09:
Hi,

this is a very nice feature request, so I'd like to quote on it.

I investigated the possibilities of USB host fingerprinting in a very early decelopment stage of P4wnP1 and haven't implemented it for the following reason:

USB based fingerprinting has to be done on lower USB layers, which are aware of descriptor requests and the corresponding timings. Unfortunately this isn't possible on USB gadget basis, which abstract away the handling of the lower layers. In fact one has to modify the kernel modules which are dealing with the lower layer, in order to get to the information for OS fingerprints. 

Of course, there're application layer approaches or to be more precise neetwork based approaches (e.g. nmap OS fingerprinting). These aporoaches have shortcomings, as they rely on a network connection (not pure USB) and involve heavy host interaction.

So the latter one is perfectly possible, as it would be a one-liner in a custom payload, but the kernel module based approach would be the way to go.

I'm willing to integrate modified USB kernel modules which fulfill the purpose of OS fingerprinting, if somebody implements them (of course default functionality of USB gadgets has to be available, too, to allow P4wnP1 to work)

jcstill, 2018-03-22:
I have updated my [repo](https://github.com/jcstill/P4wnP1_OS_Recognition) and added the script that I have been working on...
It seems like a lot but it is mostly just formatting text for output. lol
Anyways, it basically captures USB traffic on a given port and writes the data to a file for another script to come and analyse. 

Big WIP with some bugs and any help would be appreciated. Maybe @Swiftb0y would like to join in on this?
keep in mind that I go to school full time and work part time, so I don't really have a ton if time to work on this (which is why it took so long for so little haha). 

Also, I had some trouble getting this to work with the P4wnP1 (as of now it works just on the pi itself). Maybe @mame82 or someone could point me in the right direction?

Swiftb0y, 2018-03-23:
I’ll try to find some time in the next couple of weeks. I think this should definitely be integrated into P4wnP1 if it can be done without slowing down the Pi to much. DM me on Twitter (@Swiftb0y_DE) when you can.

mame82, 2018-04-07:
Progress on this? Haven't time to implement sth. like this myself!

Swiftb0y, 2018-04-07:
While jcstill's code looks promising, It doesn't work well with the P4wnp1 internals. He's at the point where he doesn't know what to look for because he has to figure out how the signature of the USB "handshake" correlates to an operating system. So in my opinion it would be best to add this as a payload Which collects the relevant data which people can submit of they want to contribute to p4wnp1.

jcstill, 2018-04-10:
@mame82 
Yes that is basically the point where I am at. I also havn't worked on it a lot because of school, work, and just plain procrastination. 

Like @Swiftb0y said, it doesn't seem to work well with P4wnP1 yet... and I am basically at a stalemate bc I don't have enough data to determine if there even is enough of a diffrence. This week I will work on getting it to work better with the P4wnP1 and maybe you could implement the code into a payload to capture usb data for me to go through and analyse.

Ideally how I would like the data:
-folder with os name (not tons of info, just like "WinXP" ro something)
-capture file inside folder
-file with all the data on os and hardware info (if possible)

However like I said, the capture script is not at a point where it can be implemented and I will work on it this week

jcstill, 2018-07-31:
After taking a break, I am thinking about revisiting this idea. I have had time to take some extra programming classes, so I am confident that I would be able to implement this. I have a friend from one class who is definetly interested and would help.

Since I want this to be as accurate and lightweight as it can be, I am going to code the payload in c++ (which I am now a little more comfortable with than before). This will also give me more accsess to hardware-specific things that I may need.

@mame82, I would be interested in helping with the kernal work if you are sitll down with implementing it on P4wnP1.

@Swiftb0y

mame82, 2018-07-31:
I'm currently working on a new P4wnP1 release, which is written in Go. When it comes to kernel relevant parts (like modifying modules, of course, C is the language of choice). For user space code, I'd definitely prefere Go - as wrapping external binaries is a pain.

So it depends, if your solution is a stand-alone binary, you're good to go with C. If the result is sth. which integrates into P4wnP1's boot or USB management process i, it is unlikely that I integrate it into the upcoming release.

jcstill, 2018-07-31:
Wow didn't see an entirely new relase comming... that's great!
The fingerprinter would be a stand-alone binary that requires a modified kernal to support the low-level USB stuffs. And I havn't even started writing code yet so an upcoming release wouldn't be feasible

Swiftb0y, 2018-08-06:
I don't think I can be of any help in particular here. I am currently busy with my other projects and hobbys, but I'm trying to learn Go in parallel so I can rewrite the wiki as soon as the new version gets released (maybe even sooner if I could get my hands on an alpha version before the official release; @mame82).