RoganDawes/P4wnP1 #211: Which image to use?

thenicnic, 2018-03-17:
Hi people,

I'm getting crazy. I'd like to use P4wP1 for an important project but I don't manage to get it run on my Pi Zero W.
Which version of raspbian lite do you recommend right now?
I tried raspbian_lite-2017-09-08 and raspbian_lite-2017-07-05 but these weren't working (USB recognize error in Windows 10). Should I try the brand new version from March 13? Or an old one raspbian_lite-2016-11-29 with kernel 4.4.x? I already downloaded this version but I'm having wifi troubles with this one. 

Would be nice to tell me the way I should take. 

Thank you,

nicnic

Six-Six-Zero-One, 2018-03-17:
The newest version of Raspbian Lite worked fine for me.

JudeAshly, 2018-03-19:
Newest version Raspbian Lite not working for me. @Six-Six-Zero-One can you explain how you did it

Six-Six-Zero-One, 2018-03-19:
I think I just flashed it with Etcher, what OS are you on and what flashing tool are you using?

JudeAshly, 2018-03-19:
2018-03-13-raspbian-stretch-lite and flashed with etcher

JudeAshly, 2018-03-19:
I created wpa_supplicant file in boot partition to connect pi 0 w to my wifi network. Will this clash with the installation process?

Six-Six-Zero-One, 2018-03-19:
Shouldnt do, I also made a wpa supplicant file and it ran fine. You havent modified anything else? Im sure youve already tried doing a clean install but do it again and dont do anything but flash raspbian. If that still doesnt boot then its the pi, not you.

Remember to check for the LED signals as well.

HitItAll, 2018-03-19:
I think I know what's wrong, after you are done setting up P4wnP1 have you
tried to revert the wpa_supplicant file back to how it it was? It takes
priority over the P4wnP1 settings. Once you set it back maybe things will
work again?

thenicnic, 2018-03-19:
Hi guys,

thanks for your feedback. My environment is running now - great stuff! Props to the developer.

What I did:

1. Downloaded the most updated version from https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2018-03-14/2018-03-13-raspbian-stretch-lite.zip

2. Bought a new micro SD card (the old one died yesterday and 2 GB weren't enough), now I got a normal 32 GB _SanDisk Ultra_

3. Formatted with _SD Formatter_

4. Flashed the image with _Win32DiskImager_

5. Put my _wpa_supplicant.conf_ and the _ssh_ file on the boot-partition (both not neccessary but useful)

6. Did the recommended install steps:

> sudo apt-get update && sudo apt-get -y install git
> cd /home/pi
> git clone --recursive https://github.com/mame82/P4wnP1
> cd P4wnP1
> ./install.sh

Note: Don't use _screen_ for background installation. I assume this was my fault until now (or the corrupted memory card).

7. Just _sudo halt_ and everything worked fine after re-plugging in the Pi

If somebody's interested in an ready-to-use-image, just tell me. I will upload my one for you (this will take a while because I only got 2 Mbps upload) bandwith.

Thank you and best regards

Nic

JudeAshly, 2018-03-20:
Yes it worked for me.I skipped the line apt-get update.

mame82, 2018-04-07:
Please use the ready-to-go image provided here https://github.com/mame82/P4wnP1/releases

closed 2018-04-07 by mame82