RoganDawes/P4wnP1 #212: The download function only grabs individual files.

Six-Six-Zero-One, 2018-03-17:
I've been trying to find a way to download entire folders to a USB drive or just to the P4wnP1's storage, the download function only works for files afaik, is that correct? If so are there any methods for downloading or copying folders from the target computer? (Preferably to a USB drive)

Thanks.

HitItAll, 2018-03-18:
The only way I can think of is if you compress all files in a directory
then download the ZIP. See this link
https://stackoverflow.com/questions/17546016/how-can-you-zip-or-unzip-from-the-script-using-only-windows-built-in-capabiliti
. One of the answers shows hot to call powershell and compress these files,
then you would issue the download command or, move the ZIP to a USB drive.
Hope this helps

Six-Six-Zero-One, 2018-03-18:
Yeah that will do, thank you.

closed 2018-03-18 by Six-Six-Zero-One