RoganDawes/P4wnP1 #215: cat: /tmp/dnsmasq.leases: No such file or directory

ImArtisann, 2018-04-05:
Been getting this error a lot lately on the Pi 0 W tried reinstalling it for the 4th time now dont know what im doing wrong here
[image]

mame82, 2018-04-07:
Seems to be an issue during installation of dnsmasq. Could you check if the binary is present ?

ImArtisann, 2018-04-07:
The binary isnt present

Swiftb0y, 2018-04-07:
I'd try a reinstall and pay close attention for error when running the installer script.

ImArtisann, 2018-04-07:
What I did to fix this error I ran the line 69 before running ./install.sh
`sudo apt-get -y install dnsmasq git python-pip python-dev screen sqlite3 inotify-tools hostapd autossh bluez bluez-tools bridge-utils ethtool policykit-1 tshark tcpdump iodine`
That seemed to fix the error i was having and it now works

mame82, 2018-04-07:
Please use the ready-to-go image provided here https://github.com/mame82/P4wnP1/releases

closed 2018-04-07 by mame82

piotrek442, 2018-04-12:
Thanks to you mame82 it works with ready-to-go image !
I've got the same error "cat: /tmp/dnsmasq.leases"

Yes !