RoganDawes/P4wnP1 #222: Manually start Payload (LockPicker)

anonymouz4, 2018-04-14:
Is there any way to start the LockPicker Payload manually?
I want to leave my Pi Zero plugged into the machine for a longer time, ssh into the Pi, and then when I want to, start the Hash Grabber.
Any Ideas?

mame82, 2018-04-15:
There's no support for this. To be honest, what you're realy requesting is USB enumeration on-demand, which is the aim of the rework branch, but I won't change LockPicker, because:
A) Typical usecases have drive-by fashion
B) The issue the payload relied on is fixed

anonymouz4, 2018-04-15:
Could I then just disable the Payload, plug it in, and when I need it, I enable the payload and reboot?

Is a payload enabled when you save the setup file or after a reboot?

Swiftb0y, 2018-04-15:
1) No, because the payload has to be run on startup (at least currently, this will probably change in the future). 

2) if you have a patched machine, yes. But some people (especially companies) don't patch regularly.

anonymouz4, 2018-04-15:
When did Microsoft patch this?

mame82, 2018-04-15:
https://github.com/mame82/P4wnP1/blob/master/writeup_lockpicker.md

closed 2018-04-15 by mame82