RoganDawes/P4wnP1 #223: Disable P4wnP1 or use default system wifi setting

anonymouz4, 2018-04-14:
Before installing P4wnP1, I used the default `/etc/wpa_supplicant/wpa_supplicant.conf` to define the accesspoint it should connect to.
Now that I've installed P4wnP1, nothing works. If I plug in my ethernet adapter, it doesn't use it as connection.

Is there a way to disable P4wnP1 temporarily?

anonymouz4, 2018-04-15:
I've done many changes, so I can't just use a fresh sd, like DynDns and so on. To get all of those things working takes a lot of time so a solution to enable ethernet or to uninstall P4wnP1 would be better

mame82, 2018-04-15:
@anonymouz4
If you'd like to bring up a ethernet based OpenVPN endpoint, there're solutions for this. But P4wnP1's goal is a different one. 
I suggest using a proper distro for that.

Thanks @Swiftb0y for trying to help in this, anyway.

I'm closing this issue, as it hasn't any relation to this project or features provided by it, nor could I spot a real issue with the project in this conversation.

closed 2018-04-15 by mame82

Swiftb0y, 2018-04-15:
Note: even permanently uninstalling P4wnp1 is difficult. So if you want your VPN server to be stable, reinstall raspbian and setup the VPN again.

mame82, 2018-04-15:
As already pointed out:
Different usecase, different sdcard.
That's why the PCB provides SD instead of fixed flash

anonymouz4, 2018-04-15:
@mame82 It would be nice if you could add in the ReadMe File that you cam't uninstall, once installed?

anonymouz4, 2018-04-15:
My idea was to get both things working on one sd. Also everything else exept the ethernet part is working so I thought there might be a workaround for this.

Swiftb0y, 2018-04-15:
Again, even if you can find a magic workaround, it probably won't be reliable at all.