RoganDawes/P4wnP1 #224: Can't create accesspoint /w Pi Zero W

anonymouz4, 2018-04-18:
Her's the output of `sudo journalctl -u P4wnP1.service`

```
-- Logs begin at Wed 2018-04-18 11:21:55 UTC, end at Wed 2018-04-18 11:23:08 UTC. --
Apr 18 11:22:00 MAME82-P4WNP1 systemd[1]: Starting P4wnP1 Startup Service...
Apr 18 11:22:01 MAME82-P4WNP1 bash[176]: =================================== P4wnP1 startup ===========================================
Apr 18 11:22:01 MAME82-P4WNP1 bash[176]: P4wnP1: Init LED control...
Apr 18 11:22:01 MAME82-P4WNP1 bash[176]: P4wnP1: Loading config ...
Apr 18 11:22:03 MAME82-P4WNP1 bash[176]: P4wnP1: Initializing USB gadget ...
Apr 18 11:22:04 MAME82-P4WNP1 bash[176]: crw------- 1 root root 243, 0 Apr 18 11:22 /dev/hidg0
Apr 18 11:22:04 MAME82-P4WNP1 bash[176]: crw------- 1 root root 243, 1 Apr 18 11:22 /dev/hidg1
Apr 18 11:22:04 MAME82-P4WNP1 bash[176]: crw------- 1 root root 243, 0 Apr 18 11:22 /dev/hidg0
Apr 18 11:22:04 MAME82-P4WNP1 bash[176]: crw------- 1 root root 243, 1 Apr 18 11:22 /dev/hidg1
Apr 18 11:22:04 MAME82-P4WNP1 bash[176]: P4wnP1: Checking for WiFi capabilities ...
Apr 18 11:22:04 MAME82-P4WNP1 bash[176]: P4wnP1: Seems WiFi module is present !
Apr 18 11:22:06 MAME82-P4WNP1 sudo[284]:     root : TTY=unknown ; PWD=/sys/kernel/config/usb_gadget/mame82gadget ; USER=root ; COMMAND=/sbin/ifconfig wlan0 up
Apr 18 11:22:06 MAME82-P4WNP1 sudo[284]: pam_unix(sudo:session): session opened for user root by (uid=0)
Apr 18 11:22:06 MAME82-P4WNP1 sudo[284]: pam_unix(sudo:session): session closed for user root
Apr 18 11:22:06 MAME82-P4WNP1 bash[176]: Try to find WiFi Hauer123
Apr 18 11:22:06 MAME82-P4WNP1 sudo[294]:     root : TTY=unknown ; PWD=/sys/kernel/config/usb_gadget/mame82gadget ; USER=root ; COMMAND=/sbin/iwlist wlan0 scan essid Hauer123
Apr 18 11:22:06 MAME82-P4WNP1 sudo[294]: pam_unix(sudo:session): session opened for user root by (uid=0)
Apr 18 11:22:07 MAME82-P4WNP1 sudo[294]: pam_unix(sudo:session): session closed for user root
Apr 18 11:22:07 MAME82-P4WNP1 bash[176]: Network Hauer123 not found
Apr 18 11:22:07 MAME82-P4WNP1 bash[176]: P4wnP1: Join present WiFi didn't succeed, failing over to access point mode
Apr 18 11:22:07 MAME82-P4WNP1 dnsmasq[317]: started, version 2.76 DNS disabled
Apr 18 11:22:07 MAME82-P4WNP1 dnsmasq[317]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify
Apr 18 11:22:07 MAME82-P4WNP1 dnsmasq-dhcp[317]: DHCP, IP range 172.24.0.2 -- 172.24.0.100, lease time 5m
Apr 18 11:22:07 MAME82-P4WNP1 dnsmasq-dhcp[317]: DHCP, sockets bound exclusively to interface wlan0
Apr 18 11:22:07 MAME82-P4WNP1 bash[176]: USB OTG off, going on with P4wnP1 boot
Apr 18 11:22:07 MAME82-P4WNP1 bash[176]: P4wnP1: ... USB gadget initialized
Apr 18 11:22:07 MAME82-P4WNP1 bash[176]: P4wnP1: Initializing Ethernet over USB...
Apr 18 11:22:07 MAME82-P4WNP1 bash[176]: CDC ECM or RNDIS active. Check which interface has to be used via Link detection
Apr 18 11:22:08 MAME82-P4WNP1 systemd[1]: Started P4wnP1 Startup Service.

```
Any Ideas?

Swiftb0y, 2018-04-18:
Can we get the hostapd and dnsmasq logs?

anonymouz4, 2018-04-18:
How do I get those?

Swiftb0y, 2018-04-18:
`sudo journalctl -u hosted.service` and the same with dnsmasq

anonymouz4, 2018-04-19:
`sudo journalctl -u hostapd.service`:
```
Apr 19 10:02:28 MAME82-P4WNP1 systemd[1]: Starting LSB: Advanced IEEE 802.11 management daemon...
Apr 19 10:02:31 MAME82-P4WNP1 systemd[1]: Started LSB: Advanced IEEE 802.11 management daemon.
```

`sudo journalctl -u dnsmasq.service`:
```
-- No entries --

```

anonymouz4, 2018-04-23:
@Swiftb0y Any Idea?

anonymouz4, 2018-04-23:
I've reseted my pi completely with current image, enabled ssh via ssh file, booted up, did **only** run `passwd` command and then the install commands of p4wnp1.

Installed normally, only errors I saw later in the log were:
```
update-rc.d: error: cannot find a LSB script for ntp
update-rc.d: error: cannot find a LSB script for dnsmasq
```
Rebooted and **NO** Accesspoint

THIS IS NOT ONLY A PROBLEM FOR ME APPARENTLY, PROBABLY FOR EVERYONE THAT USES THE CURRENT SCRIPT/IMAGE!

anonymouz4, 2018-04-23:
I've an other ad card with my vpn server where wifi in client mode works perfect right now. On this device I've also been able to create an accesspoint once but I've erased the disk since that

anonymouz4, 2018-04-23:
I think I found the problem:
The script tries to apt-get install hostapd & dnsmaq but it fails bc apt-get doesn't know them. The easiest way to fix this would be to insert apt-get update before the script starts

closed 2018-04-23 by anonymouz4

Swiftb0y, 2018-04-24:
Good to know. I'll add this to the troubleshooting page on the wiki.