RoganDawes/P4wnP1 #230: not an issue: Custom P4wnP1 USB-Stick like enclosure

careyer, 2018-04-22:
Hi @mame82, Hi All,

I love my little P4wnP1 fellow so much that I deceiced to build a custom enclosure for it. Together with a [USB-A Addon Board V1.1 USB Connector Expansion Board For Raspberry Pi Zero / Zero W](https://www.banggood.com/USB-A-Addon-Board-V11-USB-Connector-Expansion-Board-For-Raspberry-Pi-Zero-Zero-W-p-1234866.html?rmmds=myorder&cur_warehouse=CN) it now sits in a CNC milled POM case with a nice Acrylic cover so that one can see the nice guts. Handling is with a standard USB-Stick - works perfect!

Thought It might be interesting to share.

[image]
[image]
[image]
[image]
[image]

Cheers!
Keep up the excellent work!

mame82, 2018-04-22:
Looks amazing... nice work. Posted it on Twitter

harshil2105, 2018-06-01:
Buy such cases on AliExpress

ingochris, 2018-06-01:
Looks good!

virimmune210393, 2022-02-24:
> Hi @mame82, Hi All,
> 
> I love my little P4wnP1 fellow so much that I deceiced to build a custom enclosure for it. Together with a [USB-A Addon Board V1.1 USB Connector Expansion Board For Raspberry Pi Zero / Zero W](https://www.banggood.com/USB-A-Addon-Board-V11-USB-Connector-Expansion-Board-For-Raspberry-Pi-Zero-Zero-W-p-1234866.html?rmmds=myorder&cur_warehouse=CN) it now sits in a CNC milled POM case with a nice Acrylic cover so that one can see the nice guts. Handling is like with a standard USB-Stick - works perfect!
> 
> Thought It might be interesting to share. [image] [image] [image]
> 
> Cheers! Keep up the excellent work!

Hello! I really liked the case. I wanted to request you for the cover of the case (the acrylic cover). May I have stl files or dimensions for those? Where did you order those from? I was really interested in ordering on. Would be great if you can help!

Thanks,
Siddhesh

careyer, 2022-02-24:
Thanks @virimmune210393 for your interest.
This is all custom milled from Polyoxymethylene (POM) and Acrylic. Nothing stock here sorry ;-)