RoganDawes/P4wnP1 #235: Can i connect Pi as USB Wifi Adapter ?

itsStar, 2018-05-06:
I wan to know if i can connect the P4wnP1 as **USB Wifi Adapter** to Kali Linux so i can perform some attacks on network ?

mame82, 2018-05-06:
P4wnP1 isn't eyactly a WiFi adapter. But yes, you can craft a payload to route (or NAT/Masuerade) from WiFi to USB Ethernet or the other way around.
You basically have 3 interfaces (USB ethernet, WiFi and bluetooth if PAN is enabled) and can route traffic as you like.

itsStar, 2018-05-07:
That's a good news :) 
Actually i am creating an All in One solution with P4wnP1 i have created a bootable bin file with puppy linux to booth the PC from Pi as Live USB. now i am trying to use P4wnPi as USB wifi Card so i have no need to take my Alpha with me just took the P4wnP1 with me and it will do all my needs :)
Currently Pi is acting as 

1: USB HID (Keyboard)
2: USB HID (Mouse)
3: USB LAN Interface
4: USB Live bootable
5: Wifi Router With runing linux
6: USB Wifi card inProgress :)
7:.............

itsStar, 2018-05-07:
And here is what i have found during my research 
https://build.nethunter.com/nexmon-pi-builds/kali-0.5-rpi0w-nexmon-P4wnp1.img.xz