RoganDawes/P4wnP1 #237: Question: Can I connect to the Pi over a network using payloads?

IronPhreak, 2018-05-14:
I've setup my Pi to automatically connect to a mobile WiFi hotspot, in the hopes that I can use it as extra range between me and the target.

I.E plug the pi into the target computer, place the hotspot at some distance away, then I connect to the hotspot with my laptop and ssh into the Pi.

Is it possible to achieve this?

mame82, 2018-05-14:
I doubt that you're able to reach your phone via Internet, because it is likely running behind a NAT gateway.
But your phone could reach the Internet and thus your Pi is able to connect to an external SSH server (Pi is the client).
This SSH connection, again, could be used to forward a remote port of the external SSH server to Pi's local SSH server (SSH remote port forwarding). Ultimately P4wnP1's SSH server could be accessed from the Internet SSH server (and relayed by chaining SSH session etc.)
The only disadvantage: Your whole forwarding setup dies, if your (mobile) SSH session gets interrupted. This could be mitigated with autossh.

In essence you end up with this:

https://github.com/mame82/P4wnP1/blob/master/payloads/hid_backdoor_remote.txt

IronPhreak, 2018-05-14:
Thanks for your response,

Sorry I might not of made myself clear. I'm not looking to connect via the internet.

I'm looking to effectively extend the WiFi range.

Pi zero will connect to the hotspot, I'll place the hotspot a a distance away from the target, lets say 10 meters.

Then, with my laptop I go even further away so I'm out of range of the Pi, but I can connect to the hotspot. No mobile internet used, just my laptop and the Pi connecting to the hotspot which is acting as a router/wifi range extender

harshil2105, 2018-06-01:
Use a high power external WiFi adapter on your computer to increase the possibility of connecting to your raspberry pi zero w hotspot with more range. I hope you understand