RoganDawes/P4wnP1 #241: On some computers, the frontdoor doesn't work?

AgeOfMarcus, 2018-05-22:
I've yet to understand why, but on some computers, the frontdoor just waits with a ">> " prompt on powershell after typing out the initial payload. 

It's not to do with the version, I'm pretty sure, but I have no clue. Has anyone else seen this? Is there a known solution? I'd be happy to provide any screenshots or anything you need.

harshil2105, 2018-06-01:
That's the sigh for shell command line. Try giving some commands or try to Reinstall the whole P4wnP1 attacking tool.

AgeOfMarcus, 2018-06-02:
Been there, done that. If I enter a semicolon it tries to execute the code
and gets an error. When I can get back to my computer I'll do it again and
post the result.