RoganDawes/P4wnP1 #243: Help @ Installation

hanshaze, 2018-05-31:
I have installed a fresh downloaded Raspbian Lite on my Pi Zero W, updated it, upgraded it, done Kernel Update so far and the on the go function i have also "unlocked"......

so my next step should be to install P4wnP1......
I installed git and downloaded the repo......#recursive

Then i used nano to customize the setup.cfg to this --->

[image]

At this time my "/etc/network interfaces" looks like this --->

[image]

Now, the problem is, after a reboot i cant see a AP like described in setup.cfg....
I can connect to the Pi via 172.16.0.1 but i dont have internet connection anymore to install stuff....

At least i know, i have to switch setup.cfg to this to get "theoretically" internet access --->

[image]

I also customized the point of SSID/WPA in the setup.cfg file and the payload-file either....

Long story short...
WHY I DONT HAVE INTERNET ACCESS???

FSEC-DALMASIAN, 2018-06-01:
did you under the wifi client part of the setup.cfg did you fill in your correct ssid and psk of the network that you want to connect to and did you make sure to change false to true?

FSEC-DALMASIAN, 2018-06-01:
[image]

harshil2105, 2018-06-01:
This whole attacking system is offline once installed. You have to make changes in the config file to connect to your know network connection like to your WiFi. Feel free to contact me if you face any more problems.