RoganDawes/P4wnP1 #245: Can i add and rgb led to a few gpio pins to allow scripts to interact like the bashbunny

FSEC-DALMASIAN, 2018-06-01:
Can i add and rgb led to a few gpio pins to allow scripts to interact like the bashbunny
if possible can anyone help me write up the code.
i have already added code allowing a switch to change payloads while off and i also reduced the boot up time by half simply with cleaning out unnecessary packages and startup scripts.

harshil2105, 2018-06-01:
NOT AT THE MOMENT. BUT you can use NumLock and shift buttons. To change payloads like bashbunny

FSEC-DALMASIAN, 2018-06-01:
what will i need to do to allow it, dont i just need at emulate act leds through gpio then edit the led blink script for the p4wnp1 to work with it?

harshil2105, 2018-06-01:
You might be able to.....but there risk of damaging rpi0. You can use it maybe,let us know if it works.

FSEC-DALMASIAN, 2018-06-02:
I have already programed rpi0 as gpio 4 and that works well

closed 2018-06-02 by FSEC-DALMASIAN