RoganDawes/P4wnP1 #246: DNS not working via usb0 (First login via SSH)

CorpOfHack, 2018-06-16:
Hello, need help or some advice with enabling DNS on usb0 after 1st login...
I downloaded **last version of Debian Scretch Lite image** from here https://www.raspberrypi.org/downloads/raspbian/.
Flashed SD card with **Etcher**. Enabled traffic forwarding , seted up the ip address of host machine and connected via SSH to Raspberry Pi zero w. Then add iptables rules for nat masquerading. 
Added **8.8.8.8 to /etc/resolv.conf**
Added route to 169.256.x.1 (host machine)
Pings to any IPv4 wokrs fine
but any DNS request not working ((( 
Where can be the issue? I played with /etc/network/interfaces after connect
and even when flashed. But still can not get any right DNS settings
tried to add static ip to usb0 interface, but I added some changes there , I am losing usb0 interface on my host. 

Please help anybody to solve this issue. Thanks.
[image]
[image]

Swiftb0y, 2018-06-16:
I had the problem as well but I fixed it by directly setting the DNS Server via DHCP. The config gets generated in boot/init_usb_ethernet.sh

CorpOfHack, 2018-06-16:
Should I find this script on /boot on host or raspberry machine???
[image]

CorpOfHack, 2018-06-16:
I neet run the command **apt-get update && apt-get install git** 
but have a proble

I have an issue with first connection to raspberry...

Swiftb0y, 2018-06-16:
I don't quite understand what you are trying to accomplish. Which machine do you want to update? Does your Pi has internet access? How are you connected to  the pi?

CorpOfHack, 2018-06-16:
Thank a lot for reply)
Actually problem is not with P4wnP1 , I can,t install it because can't do simple
**apt-get install git** , 
I connect to my raspberry  Pi via ssh by usb0 interface from Linux host machine.
Add route on Pi machine something like :
`sudo route add -net default gw 169.256.x.1` ( on Pi machine)
then enable forwarding and accept nat masquerading from usb0 to wlan0
`iptable -t nat -A POSTROUTING -o wlan0 -j MASQUERADE` (hostmachine
enable forwarding 
`echo "1" > /proc/sys/net/ipv4/ip_forward`
then I able ping external source only by IPv4 , but not abble to do 
`ping google.com`
only can 
`ping 8.8.8.8`
Need to fix the DNS problems before installing P4wnPi
`/etc/resolv.conf` not solves this issue.
With DNS issue I not able to do simple `apt-get update && apt-get install git`
Maybe somebody had same issues please help to solve this first step with Raspberry Pi.

Swiftb0y, 2018-06-16:
Ohhh. I've never gotten the internet sharing to work. I've always connected my Pi0W to my home wifi and got internet that way.

CorpOfHack, 2018-06-17:
Thanks. Will try to. I just wanted to do steps as Mame82 :)