RoganDawes/P4wnP1 #249: Feature Request

itsStar, 2018-07-06:
Can we have a payload to do an eval portal like fluxion ?
It can be done in 2 phases 1: capture handshake 2: run portal to phish the password :)
What do you say about it ?

Swiftb0y, 2018-07-06:
I'd just create a wrapper Payload around fluxion (idk if fluxion requires interactive  user intervention tho).

itsStar, 2018-07-06:
We can do that via a web portal something like ip/admin to do the interactive user intervention OR just hard code a SSID in payload file and then run the payload so that @ least we can have a working functionality.

itsStar, 2018-07-06:
In this way we can have a portable device to run eval portal and we can connect that with mobile phone to do the administration tasks

itsStar, 2018-09-13:
Do you have any plane to release the new work ?

Swiftb0y, 2018-09-13:
I suggest you follow [him](https://twitter.com/mame82) or the [P4wnP1 Twitter](https://twitter.com/P4wnP1) for updates (which he posts every other milestone) or [me](https://twitter.com/Swiftb0y_DE) (if you want to know about Wiki Updates).