RoganDawes/P4wnP1 #255: hid backdoor hangs at FireStage1

sircosec, 2018-07-28:
I've installed fresh from the pre-build image and run ./P4wnP1/hidtools/backdoorP4wnP1.py manually as it didn't run automatically. When I run the script I get the prompt as expected but running "FireStage1" never completes and returns anything. Nothing appears to happen on the target machine either. Let me know if there are any items I can provide to troubleshoot.

Swiftb0y, 2018-07-28:
I think there are similar issues in which some debugging flags were mentioned. I can't remember them right now so you would have to dig through the existing issues to find them. It could also be possible that the script requires the currect working directory. So try running it from inside the P4wnp1 folder.

sircosec, 2018-07-28:
I was running it from the P4wnP1 folder initially.

Swiftb0y, 2018-07-28:
Try running it from inside the backdoor folder. [Ref](https://github.com/mame82/P4wnP1/blob/62b5c40d9461e4b80998c54830625d6c855833f0/payloads/hid_backdoor.txt#L58)

sircosec, 2018-07-28:
Apologies, I misspoke before. I was running it from ./P4wnP1/hidtools/backdoor/ the whole time. No change.