RoganDawes/P4wnP1 #256: hid_keyboard.txt payload will not run on mac

thewitchkingleader, 2018-07-31:
I have the Pi Zero W connected to wifi and I changed the VID and PID to those of a mac keyboard. When I plug it in via USB OTG the mac says a new network device needs to be configured. Also, the script will not run. I have enabled USB OTG and I have turned the values in setup.cfg and the payload for the ECM and RNDIS to false.

thewitchkingleader, 2018-07-31:
Also, is there a way to disable onKeyboardup so you can just set a delay in the script at the beginning? The reason I ask is because I don't know if it is sending the keyboard up command to trigger the script.

ohbriansung, 2019-03-29:
https://github.com/ohbriansung/usb_rubber_ducky/tree/master/pi_zero_ducky