RoganDawes/P4wnP1 #257: brctl error on boot; Cannot find device "pan0"

Robert-K, 2018-08-01:
> Aug  1 20:44:44 0x1DEADB01 systemd[1]: Started Hostname Service.
Aug  1 20:44:44 0x1DEADB01 bash[617]: sudo: brctl: command not found
Aug  1 20:44:45 0x1DEADB01 bash[617]: sudo: brctl: command not found
Aug  1 20:44:45 0x1DEADB01 bash[617]: sudo: brctl: command not found
Aug  1 20:44:45 0x1DEADB01 bash[617]: SIOCSIFADDR: No such device
Aug  1 20:44:45 0x1DEADB01 bash[617]: pan0: ERROR while getting interface flags: No such device
Aug  1 20:44:45 0x1DEADB01 bash[617]: SIOCSIFNETMASK: No such device
Aug  1 20:44:46 0x1DEADB01 bash[617]: Cannot find device "pan0"
Aug  1 20:44:46 0x1DEADB01 bash[617]: sudo: dnsmasq: command not found
Aug  1 20:44:46 0x1DEADB01 systemd[1]: Created slice User Slice of pi.
Aug  1 20:44:46 0x1DEADB01 systemd[1]: Starting User Manager for UID 1000...
Aug  1 20:44:46 0x1DEADB01 systemd[1]: Started Session c1 of user pi.
Aug  1 20:44:46 0x1DEADB01 bash[617]: sudo: bt-adapter: command not found
Aug  1 20:44:46 0x1DEADB01 bash[617]: sudo: bt-agent: command not found
Aug  1 20:44:47 0x1DEADB01 bash[617]: sudo: bt-adapter: command not found
Aug  1 20:44:47 0x1DEADB01 bash[617]: sudo: bt-adapter: command not found
Aug  1 20:44:47 0x1DEADB01 bash[617]: sudo: bt-adapter: command not found
Aug  1 20:44:47 0x1DEADB01 systemd[1]: Started P4wnP1 Bluetooth NAP service.
Aug  1 20:44:47 0x1DEADB01 systemd[1]: Reached target Bluetooth.

I installed P4wnP1 on a Pi Zero W running a fresh Stretch Lite install (with a monitor attached).
These lines appear at/after every boot. The interface pan0 (obviously) isn't created. I guess that's a problem?
brctl seems to be the root of all evil. Apparently it's part of bridge-utils, which isn't installed.
Am I correct? Should I be worried?

mame82, 2018-08-03:
You can find the image right here, under releases.

Robert-K, 2018-08-06:
Damn, how could I miss that?? Thanks, closing this now.

closed 2018-08-06 by Robert-K