RoganDawes/P4wnP1 #258: hid_keyboard stuck modifiers in IT keyboard localization

ienaxxx, 2018-08-07:
Hi,
 first of all let me congrat with you for the excellent job: P4wnP1 rocks!

I'm playing around with HID keyboard emulation (with the default payloads, too), but i can't use special characters 'cause modifiers keys remains "stuck" during the output phase.

I set both the setup.cfg and payload "lang" parameter to be "it", and the mapping seems to be correct, for those characters that do not require modifier keys.

Let's take hid_keyboard.txt payload example. If you modify the relevant line to be:
`echo "Keyboard is running, {asd}" | outhid`

the output in notepad is something like:

`Keyboard is running, [
       `
And then a lot of strange things happens ('cause modifiers keys are "stuck") , until i double-press all the modifiers on the real keyboard, one by one. (like fast hitting CTRL-CTRL, WIN-WIN, and so on)

**Even with the default hakin9 payload, the output in the powershell is wrong (square brackets instead of curly braces, some characters not outputted, modifiers "stuck")**

I found the way mame82_gadget emulates the keyboard to be pretty similar to what is described in this article:
[https://www.rmedgar.com/blog/using-rpi-zero-as-keyboard-setup-and-device-definition](url)

...and that the localization folder under ConfigFS is still 0x0409 (us) and not 0x0410 (it). 
I also noticed that the report_descriptor files for duckhid and outhid are different, but this could be OK. :-)

What next?
Maybe no NULL character sent after each keystroke, with "it" locale?
What should I try next?

Thankyou.

mame82, 2018-08-07:
Hi,

I'm currently not reviewing issues on this P4wnP1 version (see https://twitter.com/P4wnP1/status/1026784933552041984?s=19). Additionally I'm reimplementing the HID scripting approach.

Anyway, you've included some great in-depth analysis and the problem could impact P4wnP1's successor.

So could you maybe provide some additional info, to help identifying the root cause?

- Are dead keys involved for the special keys on Italian layout.
- Are you sure that a legacy Italian keyboard isn't using 'US' strings in the device descriptors (in all cases I'm aware of, the keyboard's hardware part/silicon is the same, with different "key toppings" ... the OS handles layout translation in software. So an Italian keyboard should still be US layouted hardware "under the hood")

ienaxxx, 2018-08-08:
Hi,
 thankyou for your quick response.
I've read the Twitter stream. Great: can't wait to get P4wnP1_go! :-)

- No: there's no dead keys (well, if I understand what you meant) involved in typing special characters in Italian layout.
- It is probably like that, but i don't know... I even tried to let the us keyboard layout but obviously this doesn't work... :-)

here's an image of the italian std layout:
http://kbd-intl.narod.ru/images/it.png

mame82, 2018-08-08:
Got it.

To make a long story short, `duckencoder.py` (internally used by P4wnP1) is a port of legacy duckencoder, including its misbehaviors and errors.

The issue is here
https://twitter.com/P4wnP1/status/1026784933552041984?s=19

This ASCII char get's translated to an USB key byte and a sequence of two modifier bytes, but modifiers have to be encoded into a single byte (logical OR).

As HID reports are created out of a alternating sequence of a single key byte and a single modifier byte, adfing in a second modifier byte breaks everything (the inserted 2nd modifier byte gets interpreted as key and the next key byte as modifier ... and so on).

My new implementation won't suffer from this, which isn't really helpful, as it isn't released yet (and I haven't written an Italian layout so far).

So let me point you to a hacky idea which could solve the issue and maybe lead to a PR:

1) Manually create a combined modifier for RIGHT ALT+LEFT SHIFT in keyboard.properties (https://github.com/mame82/duckencoder.py/blob/master/resources/keyboard.properties#L13) like this `MODIFIERKEY_RIGHT_ALT_AND_LEFT_SHIFT = 0x42`
2) Change the lines with modifier sequences in `it.properties` accordingly.
F.e. `ASCII_7B = KEY_LEFT_BRACE, MODIFIERKEY_RIGHT_ALT, MODIFIERKEY_SHIFT` gets `ASCII_7B = KEY_LEFT_BRACE, MODIFIERKEY_RIGHT_ALT_AND_LEFT_SHIFT`

I know, that's an inconvenient hack, but as I already pointed out on twitter: "DuckyScript" comes with a bunch of issues and shortcomings. I'm missing the time to fix all of them and it is much easier to write something new and more sophisticated.

mame82, 2018-08-08:
The relevant files could be found in `path to P4wnP1/duckencoder/resources/`

ienaxxx, 2018-08-09:
Yeah it seems it fixed it! Great!
I'll make other tests to be sure, but hakin9 default payload works like a charm, so i think everything should be OK now.

Again many, many thanks, even to find the time to respond to an old-version bug. :-)

mame82, 2018-08-09:
Nice, maybe you could file a PR for 'duckencoder.py', so that others could use it

ienaxxx, 2018-08-09:
Sure. Do you want me to grep all the configs searching for triple keys and substitute with the new?

Hacksturcon, 2018-08-18:
Hi guys.

As ienaxxx, i'm gratefull for this interesting and funny tool. I found the same "stuck keys" situation with curly braces in Spanish layout. I solved it modifyng the key combination in "es.properties" file to right one. 

ASCII_7B = KEY_QUOTE, MODIFIERKEY_RIGHT_ALT
ASCII_7D = KEY_BACKSLASH, MODIFIERKEY_RIGHT_ALT

Im fixing another keys who aren't right in Italian>>Spanish layout (Add missing keys, fixing weird keys and adding/deleting ISO 8891 chars....), in order to right parse base64 coded files in powershell scripts.

If you want i cant PR it in a few days.

Regards!

ienaxxx, 2018-08-29:
Since with the PR the problem is gone (tested extensively), I close this issue.
Diego.

closed 2018-08-29 by ienaxxx