RoganDawes/P4wnP1 #259: [HELP] "$p" doesnt print even if using duckhid or outhid

clancrate, 2018-09-02:
A string contains the string "$p" (without the quotation marks) and if I were to print it using duckhid or outhid the rest of the string is still printed but the characters "$" and "p" get skipped. 

so if i were to print a string using duckhid:
STRING xxx$pxxx 

It would print out xxxxxx. Same would happen if i were to use outhid.
Any workaround for this issue? Any help would be appreciated. :)

clancrate, 2018-09-02:
I'm using echo to print it out using outhid

mame82, 2018-09-02:
If you're using inline strings in bash (the payloads are interpreted in by bash) for du kyscript or the outhid, it is likely that `$p` gets interpreted as bash variable.
You can suppress bash variable expansion either by putting your source string in single quotes `'×××$pxxx'` or single character escaping `"xxx\$pxxx"`

clancrate, 2018-09-02:
Thanks 😊 I discovered a workaround yesterday but this is so much better. 

From: mame82
Sent: Sunday, September 2, 2018 9:00 PM
To: mame82/P4wnP1
Cc: clancrate; Author
Subject: Re: [mame82/P4wnP1] [HELP] "$p" doesnt print even if using duckhid orouthid (#259)

If you're using inline strings in bash (the payloads are interpreted in by bash) for du kyscript or the outhid, it is likely that $p gets interpreted as bash variable.
You can suppress bash variable expansion either by putting your source string in single quotes '×××$pxxx' or single character escaping "xxx\$pxxx"