RoganDawes/P4wnP1 #260: Kali Linux RPi0w Nexmon P4wnP1     wirelsss password error!

EastwindZL, 2018-09-02:
I downloaded   https://www.offensive-security.com/kali-linux-arm-images/ 
Kali Linux RPi0w Nexmon P4wnP1 image!

But the SSID P4wnP1 password "MaMe82-P4wnP1" is not correct.

mame82, 2018-09-02:
Please understand that I currently can't give support for the Kali image, as it isn't build by myself.

Looking into the fork of P4wnP1 or the build script, I can't spot a difference leading to this issue.

P4wnP1 fork: https://github.com/nethunteros/P4wnP1

Build script: https://github.com/offensive-security/kali-arm-build-scripts/blob/master/rpi0w-nexmon-p4wnp1.sh

So I have 3 suggestions:
- open an issue in the respective repos
- use the release image of this repo (not based on Kali nethunter)
- wait for release of the P4wnP1 successor, as I plan Kali support

EastwindZL, 2018-09-02:
Thank you very much for your answer, I wish you happiness every day.

ghosted17, 2018-09-05:
Anyone know of a fix to this? 

There seems to be updates to the script but i can't get it to work. Where do i run the script? Like what os do I need a working kali install? I tried in a stock raspbian and doesn't work.

I changed all the ssids and passwords for setup and payloads for both network-only only and wifi-connect.

 Wifi connect won't connect to my accesspoint and Network only I can see the access point come up but can't connect to it.

morningtzh, 2018-09-08:
@EastwindZL I try to edit /root/p4wnp1/setup.cfg. ssid and password is configured in it.

I edited SSID to "CMCC", "p4wnp1" disappeared on my wifi list~
 (but "CMCC" is not appear too T__T   whahahahhaha

morningtzh, 2018-09-08:
Jun 22 11:12:27 MAME82-P4WNP1 bash[176]: chown: invalid user: 'pi:pi'

I found a log, is it serious? Kali dont have user "pi", I think use "root" is better.