RoganDawes/P4wnP1 #264: Can HID backdoor attack on a linux target?

meeeo, 2018-09-21:
Lots of IoT boxes running lightweight linux OS on ARM/MIPS chips, with exposed USB ports on it.
Can I use P4wnP1 on such Linux targets?
For example, a finger print time clock, sounds HID backdoor is perfect for this, plug in P4wnP1ed PiZeroW, connect Wifi, login to P4wnP1 and got a HID communication shell ...
Can I do it?

meeeo, 2018-09-21:
I tried on my Pi 3B as a target, got P4wnP1 shell successfully, but FireStage1 are some PowerShell scripts, hope this can be used on not only windows targets, for there are huge number of Linux/Android devices running everywhere.

[image]

Swiftb0y, 2018-09-21:
Well...
The short answer is no. Not even fully fledged Desktop Distributions come with powershell installed and even If, the injected Backdoor is a binary C# blob which gets executed directly so it will only on windows. What will work are all the USB features (like HID input, Media storage and Networking between the Pi and the target). So you can use the Pi and a wifi Hotspot (like a phone) to quickly bridge air-gapped targets. The other issue with HID is that most the commands you input are blind. So you have no Idea if what you were trying to do worked. I guess most IoT devices don't even offer any input-ready shell for you to execute commands from.