RoganDawes/P4wnP1 #265: Help please

D22NK, 2018-09-23:
I install this following the install.md but when i rebooted it did not come back on. The wifi network also does not show up. 
How can i fix this?

Swiftb0y, 2018-09-23:
Try the pre-built raspbian image.

D22NK, 2018-09-23:
how do i do that?

Swiftb0y, 2018-09-23:
Look at the first three paragraphs of the Readme and use dd or Rufus (depending on your OS) to flash the image to the SD-Card.

D22NK, 2018-09-23:
i am gonna try. Thanks for helping

Swiftb0y, 2018-09-23:
Do you have an adapter to Plug the Pi0W into a display?

D22NK, 2018-09-24:
Yes

D22NK, 2018-09-24:
Thanks for the help but is working now. i was using the ip 172.16.0.1 but i have to use 172.24.0.1

Swiftb0y, 2018-09-24:
Yes. The Pi will be reachable via 172.16.0.1 when connecting to it with a cable and 172.24.0.1 when connecting via wifi. But these are default values which can be changed within the payload/setup.cfg file.

closed 2020-07-30 by D22NK