RoganDawes/P4wnP1 #274: P4wnP1 Stuck On Rainbow Splash Screen

Puffball101961, 2018-11-17:
Hello,
I recently followed the installation instructions on the docs, and after I reboot the raspberry pi it is stuck on the splash screen.

Please Help

Swiftb0y, 2018-11-17:
The installation instructions in the Docs might be outdated... (I haven't looked at them in a while). I suggest you try the pre-built raspbian image which raspbian already installed. You can find a link to it in the README.md

> Important
This repo isn't really suspended, but I'm using all of my time to work on P4wnP1's successor. The new Repo is still private, but information on progress are published via twitter, from time to time (@P4wnP1 or @MaMe82).
More important: Don't waste your time following complicated install instructions: A ready-to-go image of latest P4wnP1 version could be found on the release page: https://github.com/mame82/P4wnP1/releases (seems some of you missed it).

al7aker, 2018-12-19:
don't plug the power just plug USB.

closed 2020-08-16 by Puffball101961