RoganDawes/P4wnP1 #275: What are the conditions for hash capture?

Moe-New, 2018-11-23:
Hello everyone,I want to perform Win10_LockPicker test on win7 and win10.
I set up a simple password for win10 enterprise ltsc x86 .
Captured the hash, only succeeded once.
All subsequent attempts.The LED flashes two times. 

I continue to try on win7 professional x86,In the lock screen interface, nothing is obtained.
But when I open Internet Explorer,It immediately captured the hash.
The LED flashes three times.

please tell me,What are the conditions for hash capture?

Swiftb0y, 2018-11-23:
Well, first of all, the SMB issue has been patched by Microsoft some time ago, so it probably wont work anymore on your machine. That is also why we don't offer support for that payload anymore (it won't be included in the rework as well). If you want to read some more about the technicals aspects of the exploit, I suggest reading the [technical writeup](https://p4wnp1.readthedocs.io/en/latest/Payload-Subfolder/Windows-10-Lockpicker-Subfolder/writeup_lockpicker/) by mame82.

Moe-New, 2018-11-24:
Many people in my company are still using windows without security updates.
I need to use this payload demo to enable them to update their computers.
So, I hope to get the goal quickly instead of letting them wait for the results.

Moe-New, 2018-12-02:
Ok, thanks.

closed 2018-12-02 by Moe-New