RoganDawes/P4wnP1 #279: Server won't show up

haniashraf123, 2019-01-12:
I have installed it using the README. MD file and in the cfg file I am using hid backdoor everything is going correctly but the server doesn't show up when I reboot

yilong877, 2019-01-26:
> I have installed it using the README. MD file and in the cfg file I am using hid backdoor everything is going correctly but the server doesn't show up when I reboot

hey,I have the same problem,have you solved?

haniashraf123, 2019-01-27:
Yeah you have to use a data cable to connect and after you connect wait for some time until windows recognizes it as a keyboard it will take nearly 1 minute then connect using ssh and login

If it still doesn't connects goto the file directory
**`/P4wnP1/hidtools/backdoor/`** 

and then run this command

**`python P4wnP1.py`**

and after running that the server will show up then type
**`FireStage1 `**
and test it if the client connects you are good to go and if still not working try using a different data cable or reinstall the software.

yilong877, 2019-01-31:
It works for me,thanks!👍
And I'm testing other scripts!