RoganDawes/P4wnP1 #283: GPIO trigger

thomasoumpasis, 2019-01-31:
Hello, I was wondering if I could use a push button to execute the attack(Wifi Convert Channel) after plugging the raspberry pi to the victim machine instead of pressing NumLock 5 time quickly. I have already wired a button to the raspberry pi but I can't find a way to do this.

Swiftb0y, 2019-01-31:
GPIO isn't directly accessible from HIDScript (yet?). However, you can create a modified version of the covert channel template that executes the Covertchannel HIDScript via a triggeraction (which could be triggered by a GPIO pin). You would just have to modify the HIDScript to not wait for the numlock and instead of being run immediately, only run via the triggeraction.

Swiftb0y, 2019-01-31:
You can create your own python script that accesses the GPIO and provides an interface for bash. 
You would basically have to replace the `key_trigger` function [here](https://github.com/mame82/P4wnP1/blob/master/payloads/wifi_covert_channel/hid_only_delivery64.txt#L129).

thomasoumpasis, 2019-01-31:
Sorry for the hassle but can you please explain me what you mean by "provides an interface for bash". I understand that I have to create a script that detects when a button is pressed but I don't know how to replace the key_trigger function. Also I can't find the code for the key_trigger function so I could get a better understanding of this.

Swiftb0y, 2019-01-31:
the `key_trigger` function is basically a wrapper around the [watchhidled.py](https://github.com/mame82/P4wnP1/blob/master/hidtools/watchhidled.py). You can find it [here](https://github.com/mame82/P4wnP1/blob/master/boot/init_hid_keyboard.sh#L40). It is the "interface" I was talking about. You would basically have to create your own wrapper for the GPIO script.

The way the P4wnP1 core works is that is just all a bunch of bash scripts that get included and executed based on other bashscripts ( I recommend you take a look at [boot_P4wnP1](https://github.com/mame82/P4wnP1/blob/master/boot/boot_P4wnP1) if you want to get the full picture). So if you would like to make a `gpio_trigger` function, it would be best just to put it in a new bash script and then include that in your payload via `source <path to bash snippet that defines the gpio_trigger>`

I am not good at explaining stuff but I hope you understand what I mean. If not please just ask.

Btw. I wouldn't recommend putting too much effort into modifying this legacy version of P4wnP1 since [P4wnP1 A.L.O.A.](https://github.com/mame82/P4wnP1_aloa/) is already in its open alpha state. It is a much more solid framework to build HID based attacks on but it doesn't come with most of the handy payloads this version has (wifi cover channel, hidbackdoor, etc).

mame82, 2019-01-31:
>  but it doesn't come with most of the handy payloads this version has (wifi cover channel, hidbackdoor, etc)

Small correction:
P4wnP1 A.L.O.A. includes a WiFi covert channel (Master Template to set it up). Among other techniques, the WiFi covert channel uses a HID based covert channel to deliver the client agent.

closed 2019-02-01 by thomasoumpasis