RoganDawes/P4wnP1 #284: FireStage1 works only with nohide

Spycial, 2019-02-01:
When i try to use FireStage1 the powershell just disappears randomly but with nohide it works??

Swiftb0y, 2019-02-02:
P4wnP1 prepends a stager that moves the powershell window off the screen so it can type the long payload with anybody noticing instantly by just looking at the screen. Even though the windows isn't visible anymore, it should still be focused. It probably fails either because the window looses focus for some reason and the keystrokes just "disappear", or because some security measure (eg AV) notices that the window was moved off screen and terminates the process...
I'm just speculating though. Are you sure that the injection actually fails without `nohide`?

yassinhi, 2019-02-05:
so yeah you can just use nexmon  additions, you'll find it on github

r41nm4k3r, 2019-02-15:
I am facing the same problem. I run the Firestage1 with nohide parameter and it works just fine. How do i use the nexmon additions?