RoganDawes/P4wnP1 #286: Remote Hid Backdoor

r41nm4k3r, 2019-02-15:
I can't seem to enable the hid remote backdoor. I uncomment the proper line in setup.cfg but it doesn't seem to run when i plug it to the victim pc. 

Is there a guide with proper settings to check? 

There is a demo video in youtube that shows no other action apart from plug and unplug.

yassinhi, 2019-02-16:
trigger actions templatteeeeeeeeeee!!!!!!!!!!!!!!!!!!!!!

enable trigger actions

r41nm4k3r, 2019-02-16:
I'm not on A.L.O.A. Im running simple P4wnP1 using setup.cfg.

yassinhi, 2019-02-23:
you should use trigger action command:
fire stage one
then
fire stage two

r41nm4k3r, 2019-02-28:
Again I'm not talking about A.L.O.A. I'm talking about the P4wnP1 where you activate payloads via setup.cfg. The commands you said dont exist. 

The only command you call though the console once you eneble the payload is "Firestage1". There is no Firestage2. Have you ever run it?