RoganDawes/P4wnP1 #306: WIFI_CLIENT issue

medaminekerr, 2019-06-14:


medaminekerr, 2019-06-14:
Hello
I cannot get my P4wnP1 connected to my access point with WIFI_CLIENT option 
After boot , i check syslog and see that my acces point is detected but parsing the wpa_supplicant has failed.
Is there a way to know the reason of this failure ?

Commander-Crash, 2019-09-23:
What I had to do is choose WIFI_CLIENT in the web menu then connect with BT and open a SSH session and type in "ifconfig wlan0 <IP-ADDR>" then I had to add route by typing "sudo route add default gw <ROUTER-IP>" This got it to connect but a pain in the a$$ It would be nice if a script can be made to get this to be less of a headache in case people want to update or download from the internet. I hope this helps.

Commander-Crash, 2019-09-23:
What I had to do is choose WIFI_CLIENT in the web menu then connect with BT and open a SSH session and type in "ifconfig wlan0 <IP_ADDR>" then I had to add route by typing "sudo route add default gw <ROUTER_IP> " This got it to connect but a pain in the a$$ It would be nice if a script can be made to get this to be less of a headache in case people want to update or download from the internet. I hope this helps.