RoganDawes/P4wnP1 #307: P4wnP1 DuckyScript Opens Office

ethancawse, 2019-06-20:
Hello, I am using the HID Backdoor Payload, and whenever I use the command FireStage1, my computer keeps on opening Office, or links to sign into Office. This doesn't just happen with this specific payload, because I have tried other ones, and I get the same results. I am not sure why this is happening, and any help would be great. Here is the link in which it opens: https://go.microsoft.com/fwlink/?linkid=2044481. Thanks.

Abigdog4, 2019-06-30:
I had the same problem, It only does this on windows 10 so there are two ways to fix it, You can remove the office from the computer start GUI (This is where it keeps happening) or you can fix your ducky script where under EOF it says outhid you might have lines of code interfering with it.

Personally I use run instead of using the start menu as its faster and ensures better control. 👍

testbughub, 2019-08-14:
+1 here, and I can't figure out how to fix it.

Puffball101961, 2019-08-17:
Have the same problem, keeps opening the same link, but not office.

iiAlphaWolf, 2019-08-29:
Mine does it too I've tested it on windows 10 1903 all script open office...

Karmakip, 2019-10-17:
Ik heb ontdenkt : 
Office word geopend bij een DELAY in het duckyscript.
Office word geopend omdat de volgende combinatie word ingedrukt: GUI, SHIT, AltGr.

greyjedimaster, 2019-10-18:
Has anyone found a solution to this? Or at least know where the actual FireStage1 script is located to try and edit?

Lz1y, 2019-10-19:
me tooooooo!

DawsonMacPhee, 2019-10-28:
Same issue here... really annoying. Not really sure what to do about this...

keyiflerolsun, 2020-01-05:
> I had the same problem, It only does this on windows 10 so there are two ways to fix it, You can remove the office from the computer start GUI (This is where it keeps happening) or you can fix your [...]

how are we doing?

jamal323, 2020-01-11:
I have the same problem, but it works on a friend’s computer, it just doesn’t manage to print as powershell closes right away

TrollC, 2020-01-14:
Same error here, please provide a solution

saracm93, 2020-01-24:
Try with kali linux version https://images.offensive-security.com/arm-images/kali-linux-2019.3-rpi0w-nexmon-p4wnp1-aloa.img.xz

keyiflerolsun, 2020-01-25:
> Try with kali linux version https://images.offensive-security.com/arm-images/kali-linux-2019.3-rpi0w-nexmon-p4wnp1-aloa.img.xz

this is aloa image

ViggoGaming, 2020-01-28:
Is there any fix to this?

3id98, 2020-02-05:
Got a fix, in hidtools/transhid.py change the if-statement and everything after to this:
`if (key == 0):
	# delay code
	d = float(mod)/1000.0
	time.sleep(d)
else:
	out = chr(mod) + '\x00' + chr(key) + '\x00\x00\x00\x00\x00' + '\x00\x00\x00\x00\x00\x00\x00\x00'
	f.write(out)
f.flush()`

DonaldDucker34, 2020-04-19:
So apparently the Fix above didnt work for me because it already was in Pwnpi installed but when i searched in the Internet i found a solution the fix is in my GitIssue https://github.com/RoganDawes/P4wnP1/issues/337

kagessss, 2020-04-25:
+1,how to fix it???

DonaldDucker34, 2020-04-25:
its in my post just look at the comment above you (¬_¬ )

8arcly, 2020-06-23:
@DonaldDucker34 Really?! deleting office or changing the command when running to rundll32 is no solution, at most a workaround for those who have administrator rights on the system.
Originally the script worked on every user.

For some reason, the characters sent to the system when calling the DELAY function from ducky scripts correspond to the combination Ctrl + Alt + Shift + Win. Which Microsoft shortcut has recently assigned to the launch of Office Hub.

The payload itself, which sets up the HID channel, seems to work correctly, because it sometimes manages to launch through various combinations, e.g. quick changing the window focus of the system to PowerShell window.

The correct path seems to be to fix the DELAY behavior, e.g. signaled by 3id98 with the modification hidtools / transhid.py. 
@3id98 Could you specify what the Your fix is about? In fact, except `else:` the script in P4wnP1 seems to have this content.
You can use ` ``` ` to propertly quote whole block of code.

```
	if (key == 0):
			# delay code
			d = float(mod)/1000.0
			time.sleep(d)
			out = chr(mod) + '\x00' + chr(key) + '\x00\x00\x00\x00\x00' + '\x00\x00\x00\x00\x00\x00\x00\x00'
```

I'm trying to understand the mechanics of DELAY implementation, e.g. also in the file duckencoder / duckencoder.py

ie. in the defining part
```
if cmd == "DELAY":
                        delay = int(args)
                        result = DuckEncoder.delay2USBBytes(delay)
```
...
```
       def delay2USBBytes(delay):
                result = ""
                count = delay / 255
                remain = delay % 255
                for i in range(count):
                        result += "\x00\xFF"
                result += "\x00" + chr(remain)
                return result
```

3id98, 2020-06-23:
I think the problem was that if a delay is read, it delays but after that it pushes a key and that key opens office.
Here is the whole transhid.py:
```#!/usr/bin/python

#    This file is part of P4wnP1.
#
#    Copyright (c) 2017, Marcus Mengs. 
#
#    P4wnP1 is free software: you can redistribute it and/or modify
#    it under the terms of the GNU General Public License as published by
#    the Free Software Foundation, either version 3 of the License, or
#    (at your option) any later version.
#
#    P4wnP1 is distributed in the hope that it will be useful,
#    but WITHOUT ANY WARRANTY; without even the implied warranty of
#    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
#    GNU General Public License for more details.
#
#    You should have received a copy of the GNU General Public License
#    along with P4wnP1.  If not, see <http://www.gnu.org/licenses/>.

# Transform raw input to match output for the HID report descriptor in use
# Author: MaMe82 aka.  Marcus Mengs

import sys
import time

data = sys.stdin.read()
with open("/dev/hidg0","wb") as f:
	for i in range(0, len(data), 2):
		out = ""
		key = ord(data[i:i+1])
		if len(data[i+1:i+2]) == 0:
			continue
		mod = ord(data[i+1:i+2])
		if (key == 0):
			# delay code
			d = float(mod)/1000.0
			time.sleep(d)
		else:
			out = chr(mod) + '\x00' + chr(key) + '\x00\x00\x00\x00\x00' + '\x00\x00\x00\x00\x00\x00\x00\x00'
			f.write(out)
		f.flush()
		# no delay between keypresses (hanfled by HID gadget)
		#time.sleep(0.01)```

8arcly, 2020-06-24:
I have transferred Your version entirely to my P4wnP1 and this does not solve my problem.

It seems that we need to take a closer look at the implementation of the DELAY function on P4wnP1.
In addition, I noticed other incorrect behavior - for a simple ducky script
```
STRING aaaa
DELAY 8000
STRING bbbb
```
The execution first waits 8 seconds and then sends both strings at once.

In addition, I checked identical scripts on the original RubberDucky and there is no such problem, also with opening OfficeHub.

8arcly, 2020-06-24:
@3id98 Yes! Your solution is correct but I had to put it not in transhid.py but in DuckEncoder.py. Contains the same code snippet.

3id98, 2020-06-24:
Nice

8arcly, 2020-06-24:
The problem was with sending 'technical algorithm data' (calculated with delay2USBBytes) for the delay function to the HID device. Looks like it contained strings corresponding to pressing ctrl + alt + win + shift - probably `\xff` which for most delays was sent on every second byte.
Cutting off assignment to `out` thesee technical data, recognized with `key == 0`, by adding `else`  solves the problem.
Thanks again!

knowMe228, 2021-07-27:
> @3id98 Yes! Your solution is correct but I had to put it not in transhid.py but in DuckEncoder.py. Contains the same code snippet.

Hi, can you post the contents of DuckEncoder.py and transhid.py?

xspeed125, 2021-09-11:
> @3id98 Yes! Your solution is correct but I had to put it not in transhid.py but in DuckEncoder.py. Contains the same code snippet.

Yes worked for me as well. Thank you all

knowMe228, 2021-09-19:
> 
> 
> > @3id98 Yes! Your solution is correct but I had to put it not in transhid.py but in DuckEncoder.py. Contains the same code snippet.
> 
> Yes worked for me as well. Thank you all

can you share your duckencoder.py?

knowMe228, 2021-09-21:
this line is already present in the DuckEncoder.py, but nothing works for me, how did you achieve the result

xspeed125, 2021-09-21:
Make sure that you edit every line with the delay snippet on duck encoder. There is more than one of that snippet. It's present like 5-6 times > this line is already present in the DuckEncoder.py, but nothing works for me, how did you achieve the result

8arcly, 2021-09-21:
For me, it was enough to modify the file /home/pi/P4wnP1/duckencoder/duckencoder.py (for which there is also a symlink in /home/pi/P4wnP1/hidtools/backdoor/Duckencoder.py)

Modification around line 398 

Before:
```
                                        time.sleep(d)
                                out = chr(mod) + '\x00' + chr(key) + '\x00\x00\x00\x00\x00' + '\x00\x00\x00\x00\x00\x00\x00\x00'
                                f.write(out)
                                f.flush()
                                # no delay between keypresses (hanfled by HID gadget)
                                # time.sleep(0.01)
```
After:
```
                                        time.sleep(d)
                                else:
                                        out = chr(mod) + '\x00' + chr(key) + '\x00\x00\x00\x00\x00' + '\x00\x00\x00\x00\x00\x00\x00\x00'
                                        f.write(out)
                                f.flush()
                                # no delay between keypresses (hanfled by HID gadget)
                                # time.sleep(0.01)

```
Whole file attached:
[duckencoder.zip](https://github.com/RoganDawes/P4wnP1/files/7206027/duckencoder.zip)
It contains a few extra commented lines as a remnant from tracking the problem - around lines 44 and 220.

knowMe228, 2021-09-21:
> For me, it was enough to modify the file /home/pi/P4wnP1/duckencoder/duckencoder.py (for which there is also a symlink in /home/pi/P4wnP1/hidtools/backdoor/Duckencoder.py)
>
> Modification around line [...]

Thank you ,I'll check your version tomorrow

knowMe228, 2021-09-22:
> For me, it was enough to modify the file /home/pi/P4wnP1/duckencoder/duckencoder.py (for which there is also a symlink in /home/pi/P4wnP1/hidtools/backdoor/Duckencoder.py)
>
> Modification around line [...]

your version didn't help me
[image]

knowMe228, 2021-09-22:
> 
> 
> Make sure that you edit every line with the delay snippet on duck encoder. There is more than one of that snippet. It's present like 5-6 times > this line is already present in the DuckEncoder.py, but nothing works for me, how did you achieve the result

but I do not know how to change the other parts of delay, except for the 398 line.But for some reason it doesn't help me(

8arcly, 2021-09-22:
Have you edited the duckencoder.py file yourself or have you uploaded the one I sent?

knowMe228, 2021-09-22:
> 
> 
> Have you edited the duckencoder.py file yourself or have you uploaded the one I sent?

I used your file

8arcly, 2021-09-22:
> 
> 
> > Have you edited the duckencoder.py file yourself or have you uploaded the one I sent?
> 
> I used your file

And there have been changes to other files from previous attempts?
For example, if I remember correctly, I did not leave any changes in the transhid.pl file (additionally I upload mine - [transhid.zip](https://github.com/RoganDawes/P4wnP1/files/7212525/transhid.zip))
I have one more change in /home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py but related to a different topic (Bypass constrained mode restrictions where applicable)

I've just checked P4wnP1 on my W10 and everything works as expected.
[image]
(In the screenshot from mobile connected to P4wnP1 there is a manual start of P4wnP1.py, only because I interrupted the earlier operation, but it also starts automatically after boot-up). 

You can try one more thing, if I remember correctly at some point when modifying the duckencoder.py file - reapply symlink (ln -s) to the file.
[image]

knowMe228, 2021-09-22:
I don't know what to do to make this thing work,I did everything that was written to me,but NOTHING worked,I lost heart....

8arcly, 2021-09-22:
Does any MS Office component start when you try to execute it?

knowMe228, 2021-09-23:
> Does any MS Office component start when you try to execute it?

No,after running the execution line, powershell opens ,and then the code is executed, as a result of which it returns the error that I have already sent

8arcly, 2021-09-23:
> 
> 
> > Does any MS Office component start when you try to execute it?
> 
> No,after running the execution line, powershell opens ,and then the code is executed, as a result of which it returns the error that I have already sent

It seems that you have a different problem than that described in the OP.

knowMe228, 2021-09-28:
> > > Does any MS Office component start when you try to execute it?
> > 
> > 
> > No,after running the execution line, powershell opens ,and then the code is executed, as a result of which it returns the error that I have already sent
> 
> It seems that you have a different problem than that described in the OP.

it all worked!!! The problem was with windows itself

EthanGuy-program, 2024-07-15:
+1, I have this same issue when trying to run. The passive os for windows extension, but in the Microsoft 365 (Office) app.