RoganDawes/P4wnP1 #313: ./install.sh

Jora73, 2019-08-15:
I print ./install.sh , and pi return :
Testing Internet connection and name resolution...
...[pass] Internet connection works
Testing if the system runs Raspbian Jessie or Stretch...
...[Error] Pi is not running Raspbian Jessie or Stretch! Exiting ...
Help me plz 
I use firmware from the site https://p4wnp1.readthedocs.io/en/latest/

megah3rtz, 2019-08-23:
@Jora73 What version of raspian have you installed?

DarthAluminium, 2019-12-23:
You need to install previous version of Raspbian. From the download page of Raspbian you can only download Buster, but you need Raspbian Stretch or Raspbian Jessie.

There is you can download all Raspbian versions:
https://downloads.raspberrypi.org/raspbian/images/

So I downloaded and installed last Raspbian Stretch and P4wnP1 was successfully installed.
https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-04-09/

fortworthless, 2020-01-08:
**@DarthAluminium**, thank you. I spent a day of insanity trying to make buster work, then Kali, and on and on. I looked everywhere for that old raspbian image. I even edited the install.sh to not check the version. And now I have it, finally at 2am. 

**But I didn't give up and it works!**
[image]

closed 2022-10-21 by kg04ls