RoganDawes/P4wnP1 #319: Raspberry Pi 4 support

Googulator, 2019-10-18:
On the Pi 4, the USB 2.0 OTG controller is wired directly to the USB-C socket that also handles power, without a hub in between. The USB-A sockets are connected to a separate USB 3.x xHCI controller. It should therefore be possible to use a Pi 4 for this attack.

This needs to be tested and documented.

Besix2, 2022-06-13:
I tried it. But it already failed at the instalation. Im not the only one who tried on pi 4. But if someone gets this working it would be very cool.

LeskaHub, 2023-03-06:
I tried too on pi 4B but the network is not showing. I saw that they are many un responded bugs so I guess the dev(s) stopped working in it. Very sad.