RoganDawes/P4wnP1 #326: Can't install because of Error

DarthAluminium, 2019-12-20:
pi@raspberrypi:~/P4wnP1 $ ./install.sh
Testing Internet connection and name resolution...
...[pass] Internet connection works
Testing if the system runs Raspbian Jessie or Stretch...
...[Error] Pi is not running Raspbian Jessie or Stretch! Exiting ...

What should I do to solve this error?

closed 2019-12-20 by DarthAluminium

saintofinternet, 2019-12-21:
did you find a solution for the issue?

i am facing the same " Pi is not running Raspbian Jessie or Stretch " and i have downloaded the latest Raspbian Buster Lite ( Minimal image based on Debian Buster )

DarthAluminium, 2019-12-21:
You need to install previous version of Raspbian. From the download page of Raspbian you can only download Buster, but you need Raspbian Stretch or Raspbian Jessie.

There is you can download all Raspbian versions:
https://downloads.raspberrypi.org/raspbian/images/

So I downloaded and installed last Raspbian Stretch and P4wnP1 was successfully installed.
https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-04-09/