RoganDawes/P4wnP1 #327: Can't FireStage1 because of SetKeyboardLanguage Problem

DarthAluminium, 2019-12-23:
**P4wnP1 shell (client not connected) > FireStage1**
Starting to type out stage1 to the target...
...done. If the client doesn't connect back, check the target
keyboard layout with 'SetKeyboardLanguage'
**P4wnP1 shell (client not connected) > SetKeyboardLanguage**
Choose language by number or name:
================================

0:ca    1:de    [2:us]  3:es    4:pt    5:ch    6:tr    7:hr
8:gb    9:fr    10:no   11:cs   12:be   13:it   14:dk   15:fi
16:br   17:sv   18:si   19:ru
**Your selection or 'x' to abort: 19**
language set to 'ru'
**P4wnP1 shell (client not connected) > FireStage1**
Starting to type out stage1 to the target...
r interpreted as ASCII_72, but not found in chosen language property file. Skiping character!
Traceback (most recent call last):
  File "P4wnP1/hidtools/backdoor/P4wnP1.py", line 1346, in <module>
    p4wnp1.cmdloop()
  File "/usr/lib/python2.7/cmd.py", line 142, in cmdloop
    stop = self.onecmd(line)
  File "/usr/lib/python2.7/cmd.py", line 221, in onecmd
    return func(arg)
  File "P4wnP1/hidtools/backdoor/P4wnP1.py", line 746, in do_FireStage1
    self.stage1_trigger(trigger_type=trigger_type, trigger_delay_ms=trigger_dely_ms, hideTargetWindow = hideTargetWindow, bypassUAC=bypassUAC)
  File "P4wnP1/hidtools/backdoor/P4wnP1.py", line 466, in stage1_trigger
    self.duckencoder.outhidDuckyScript(ps_stub) # print DuckyScript stub
  File "/home/pi/P4wnP1/hidtools/backdoor/DuckEncoder.py", line 427, in outhidDckyScript
    payload = DuckEncoder.parseScript(source, self.keyboard, self.language)
  File "/home/pi/P4wnP1/hidtools/backdoor/DuckEncoder.py", line 363, in parseScipt
    result += DuckEncoder.parseScriptLine(l, keyProp, langProp)
  File "/home/pi/P4wnP1/hidtools/backdoor/DuckEncoder.py", line 165, in parseSciptLine
    result = (DuckEncoder.keyInstr2USBBytes(args, keyProp, langProp) +
  File "/home/pi/P4wnP1/hidtools/backdoor/DuckEncoder.py", line 234, in keyInst2USBBytes
    keyval = DuckEncoder.ASCIIChar2USBBytes(keyinstr, keyProp, langProp)[0]
IndexError: string index out of range
Cleaning Up...
LinkLayer: Aborting sync
**pi@MAME82-P4WNP1:~ $**

Why is this happening?

lexa715, 2021-02-23:
привет, вижу ты тоже на русском говоришь,рад увидеть земляка) Хочу спросить ты нашел решение?

DarthAluminium, 2021-02-23:
> привет, вижу ты тоже на русском говоришь,рад увидеть земляка) Хочу спросить ты нашел решение?

давно было, я уже и не помню

TehBset, 2021-03-09:
Так как решить проблему?