RoganDawes/P4wnP1 #330: when I execute the command "Firestage1" P4wnP1 opens Microsoft Office

jcldf, 2020-02-01:
when I execute the command "Firestage1" P4wnP1 opens Microsoft Office, the keyboard is configured correctly and I tested it on different computers with different operating systems (win 7 and win 10).
I've run previous versions of p4wnp1 and the attacks were successful, can you help me or indicate a previous version of the project so that I can install it on my Raspberry Pi?
[image]

lsdrfrx, 2020-02-07:
Same problem

Tech-User42, 2020-02-13:
Same issue it works 1/20 times please do something this soft is incredible !

harshil2105, 2020-03-11:
same issue!!!

DonaldDucker34, 2020-04-16:
also got that issue 😞

DonaldDucker34, 2020-04-19:
In my Post is the FIX https://github.com/RoganDawes/P4wnP1/issues/337

jcldf, 2020-04-20:
> In my Post is the FIX #337

thxxxx

DonaldDucker34, 2020-04-20:
no problem hope it worked for you :)