RoganDawes/P4wnP1 #334: Mass storage Image?

Cha01605, 2020-04-08:
Hey guys,

Im new to the P4wnP1 and just got done building one with the beboxos OLED menu code. The very first thing I was trying to do is get the passwordgrabber payload working Which I believe gets stored in the mass storage section. 

This is the specific payload im referring too: 

https://github.com/akhil1136/P4wnP1-ALOA-payloads/tree/master/PasswordGrabber

Which uses this framework:

https://github.com/AlessandroZ/LaZagne

When I go to create the mass storage device I see there is the 1 test image. When you open it there is a readme file that says:

"A helper to build flashdrive and CD-Rom images could be found at
/usr/local/P4wnP1/helper/genimg

;-) cheers MaMe82"

Where do you put in the "/usr/local/P4wnP1/helper/genimg" in order to get the helper? Im assuming this helper explains how to make a larger mass storage image?

If im understanding that payload correct. I need to make a mass storage image called P4WNP1, make the 3 folders it calls for, put all the specific files in the specific folders per the instructions and thats it. 

I just need to figure out how to make a bigger mass storage drive image that is larger than a couple MBs. Any insight would be appreciate.

SymbianSyMoh, 2020-04-09:
`genimg` is the helper itself and you can find it in `/usr/local/P4wnP1/helper/` in your P4wnP1 installation!

Cha01605, 2020-04-09:
Now what I can not figure out is what is the syntax for genimg?

Even the simple example they give is coming up with an error

genimage -i whatever -o Whatever

Gives me this error:

-bash: genimage: command not found

Thanks for the insight, Sym!

SymbianSyMoh, 2020-04-09:
Simple `./genimg -h`

Cha01605, 2020-04-09:
So what is the ./ actually doing? Im assuming that its saying run whatever file is after it? in this case, genimg?

closed 2020-04-09 by Cha01605