RoganDawes/P4wnP1 #337: Office Opens

DonaldDucker34, 2020-04-18:
when i enter firestage 1 on my pc office opens. So i thought why dont edit the firestage1 command but i dont know how or is there an fix?

DonaldDucker34, 2020-04-19:
So here is a FIX for everyone i looked trough the internet and found this :(https://superuser.com/questions/1455857/how-to-disable-office-key-keyboard-shortcut-opening-office-app) when you paste the command "REG ADD HKCU\Software\Classes\ms-officeapp\Shell\Open\Command /t REG_SZ /d rundll32" in the admin command prompt the Hotkey for office isnt working any more 😄

furiousduckling, 2020-04-21:
@DonaldDucker34  How do you go about automating this using ducky script?

DonaldDucker34, 2020-04-21:
that is the 'problem' i cant connect to a pc without entering the firestage1 command that means i cant send ducky scripts to a pc

DonaldDucker34, 2020-04-21:
but i dont know if you could edit the firestage1 command that it first enters the cmd string and then connects the idea is not that bad

furiousduckling, 2020-04-21:
Is it an issue with p4wnP1 and the image? Anyone using the ALOA image
getting the issue?

furiousduckling, 2020-04-21:
Trying to think of a way to stick it on my duckberry without any interaction with the pc itself

DonaldDucker34, 2020-04-24:
so i didnt knew what a Duckyberry is but now i do. The duckberry is an bad usb and a rubber ducky then the Duckberry should use the .duck format i think you should look at the .duck scripts from P4wnP1 and make your own for the Duckberry

furiousduckling, 2020-04-24:
I made my own but no luck. I'm not getting office open now. Just using
default I'm getting the notepad launch and then file explorer opens up
around 6 times

DonaldDucker34, 2020-04-25:
could you show me that script i think its very interesting to look at something like that

furiousduckling, 2020-04-25:
It's just the default one in p4wnp1 located here :
https://github.com/RoganDawes/P4wnP1/blob/master/payloads/hid_keyboard.txt

DonaldDucker34, 2020-04-26:
if you look in the bottom area of the script it tells the pc to type notepad.exe but why explorer opens is unknown

DonaldDucker34, 2020-04-26:
is the keyboard language equal to your keyboard language? Its a really common issue

furiousduckling, 2020-04-26:
Yeah,

That is why I am confused

DonaldDucker34, 2020-04-26:
this is getting really interesting i will look trough Pwnpi and search for a fix :)

DonaldDucker34, 2020-04-26:
> 
> It's just the default one in p4wnp1 located here : https://github.com/RoganDawes/P4wnP1/blob/master/payloads/hid_keyboard.txt
> […](#)

is it really the exact same? i had a look at the script an nothing seemed wrong it just types win+r to open the execute window and then it types notepad.exe to open the editor last it types "Keyboard is running" but where does it open the explorer? i did a bit research and found that the explorer is opened by the key combination Win+E (GUI e) so i think we should search a bit more 😄

DonaldDucker34, 2020-04-26:
and what did you actually type to start the script?

furiousduckling, 2020-04-26:
Nothing, the moment I plug the USB in, it would run

DonaldDucker34, 2020-04-26:
did you selected the right payload in the setup.cfg? (cd P4wnP1 then nano setup.cfg and on the bottom there are the payloads)

furiousduckling, 2020-04-26:
Yeah, unhashed the hid_keyboard.txt

DonaldDucker34, 2020-04-27:
> 
> 
> @DonaldDucker34 How do you go about automating this using ducky script?

to this older comment: i think it would be cool if i wrote a script and then i could execute it but i cant execute ducky scripts without connection to a PC but i also cant connect to a PC without the cmd string then i thought about an autorun.inf on a smaller usb stick (2-4gb) but since win10 you cant or windows wont execute autorun.inf because its an security leak. Would it be possible that the rpi runs the ducky script on rpi startup then pwnpi uses its HID capability and sends the cmd string without the remote shell even activated that means i cant enter things in the remote shell but theoretically it should disable the office hotkey and then i could enter FireStage1 to connect to the rpi and get an reverse shell